stratux/stratux #4: Worth implementing Flarm-compatible output for bearingless targets?

b3nn0, 2019-05-06:
In the original Stratux repository, there is a lengthy discussion from 2017 regarding Flarm-NMEA output:
https://github.com/cyoung/stratux/issues/489
aswell as initial work done:
https://github.com/cyoung/stratux/pull/560

While this seems not entirely needed these days, since most EFBs now support GDL90, I want to discuss if bringing these patches to a usable level in modern Stratux would still be worth it?
From the discussion, I take that Flarm-NMEA supports bearingless targets (which GDL90 does not) and this actually seems to be implemented in many EFBs.

- Is it still the case that SkyDemon can only work with FLARM if it is present under 192.168.1.1?
- How about using a different protocol? What does PilotAware's protocol offer and is it documented? Or SkyEcho?

Feel free to discuss...

VirusPilot, 2019-05-06:
I believe GDL90 supports bearingless targets, in my case I saw such targets in ForeFlight, connected to my Air Avionics Air Traffic device. The display method in FF though is not perfect - a number of dots that connect to a virtual circle. I haven't tested this with SD but using FLARM-NMEA with SD is much nicer - SD actually draws a complete circle. Still I believe that implementing FLARM-NMEA should not be #1 priority.

TomBric, 2019-05-11:
GDL90 in general supports bearingless targets. See this in the GDL90 spec on FAA:
"A target with no valid position has Latitude, Longitude, and NIC all set to zero. "

Skydemon should also display GDL90 bearingless targets if enabled (that says the documentation, but I have not experienced that yet, when connecting SD to my Stratux)

For me FLARM-NMEA as well has no priority ...

b3nn0, 2019-05-11:
You are absolutely right. I read in some random forum post that GDL90 wouldn't support it, but checking the spec myself confirmed what you say.
In this case, I would also say that there is no point in implementing this for now.
@TomBric: Anyway, Stratux does not report bearingless targets via GDL90, that's why you will not see them in SD.

b3nn0, 2019-07-04:
Reponeing this. I have investigated some more:
While it is possible to send GDL90 Targets without position, it is impossible to state the distance to that target in GDL90. Therefore, this information is useless for the EFB.
Interestingly, some EFBs seem to interpret the NACp value for bearingless targets and use that to draw a distance ring. That, however, is non-standard and not widely supported.

In the flarm-nmea branch, I have done some experiemnts for analyzing correlation between signal strength and traffic distance, with a self-adjusting tuning parameter that is learnt from ADS-B signals and then used to predict Mode-S signals. Flarm NMEA output works aswell (however, to make this connect to your EFB, you might have to change your Stratux IP to 192.168.1.1).

The value in the braces show the computed estimate, the one outside show the real value from the ADS-B Positions. At least for close traffic (< 15 nm), it seems to work rather well.
[image]

Some more things to consider:
- FLARM can only transmit one Mode-S target (the closest one). I think that is enough.
- Right now, one parameter is tuned via ADS-B to predict the distance of non-ADS-B traffic. This does not consider the fact that usually large aircraft have a higher transmit power than smaller ones.
This means, that with this one tuning parameter, we will usually show small aircraft too far aways and large aircraft too close. Is it worth to keep one tuning parameter for each aircraft category (Emitter_category) and use that to only predict distances for aircraft of the same category? How reliable is this Emitter_category? From what I can see, it isn't at all - maybe a bug in dump1090?

VirusPilot, 2019-07-04:
Are you proposing to use the FLARM-NMEA protocol for transmitting Mode-S targets to the EFB? If so, keep in mind that some popular EFBs only support GDL90, e.g. ForeFlight ...

b3nn0, 2019-07-04:
Yes, that is what I am proposing. Don't worry, I don't want to replace GDL90, just offer another option.
With Flarm-nmea, it is possible to transmit bearingless targets with a distance (which will be drawn as a circle in SkyDemon). With GDL90, it is not. I can transmit targets without a position, but there is no way for the EFB to know its distance (or is there any way that works for Foreflight?) I'm happy to provide the same distance info via GDL90, if there is a way to make EFBs display it.

VirusPilot, 2019-07-04:
Attached an example how ForeFlight shows Mode-S targets, no idea whether any distance is displayed:
[image]

VirusPilot, 2019-07-04:
A related observation: ForeFlight is now promoting uAvionix SkyEcho 2 with FLARM support and for a short period of time SkyEcho 2 had even implemented Mode-S targets but they removed this feature recently. I wonder how FF would have displayed such targets ...

b3nn0, 2019-07-04:
That is.. interesting. What is this foreflight connected to? Because for now, Stratux does completely drop any bearingless targets from its GDL90 stream. Only aircraft with a valid position are sent via GDL90 - and I don't see any useful way to change that - There are always mode-S targets around - without any information about their estimated distance, the EFB really can't do anything useful with it..
See https://github.com/cyoung/stratux/issues/489 for how many EFBs show the bearingless targets WITH distance estimate - but that's Flarm-NMEA stuff, not GDL90.

No idea what protocol SkyEcho2 is using, but since they removed this again, I would not bet on many EFBs supporting their implementation in the future..
According to https://forums.flyer.co.uk/viewtopic.php?t=112166:

> Bearingless targets are a feature provided by Flarm and PilotAware. Those devices pass us the aircraft id, altitude, vertical speed, estimated range (guesswork but based on signal strength; the user can learn to interpret the ring size changes over time) and even callsign.
> 
> SkyEcho doesn’t have anything like that. Its only offering is to send us the altitude of a mode C/S return as it comes in. No vertical speed, no range (estimated or otherwise), or even an aircraft id. I declined that offering.

VirusPilot, 2019-07-04:
I had FF connected to AT-1 from Air Avionics. They have implemented GDL90 recently together with Mode-S display. I could simply ask them how they did this implementation ...

b3nn0, 2019-07-04:
If you have a contact there, it would be great to know how they encode this in GDL90

b3nn0, 2019-07-21:
I just implemented something like AirAvionics in GDL90, by simulating 8 targets in a circle, which indicates the distance. Will look something like this:
[image]

Limits for traffic to be displayed are the same as for NMEA output: +-2000ft, 15km (8nm) distance.
@VirusPilot Would you mind testing this with ForeFlight?

closed 2019-08-05 by b3nn0