stratux/stratux #7: FLARM option not working

CHRE2, 2019-05-19:
Moin.

Just installed your latest version 'Stratux Europe Edition 1.5b2-eu002' and change the UDP/IP for AirNav Pro use. https://github.com/0sander/stratux_air_nav_pro
Like it very much, especially the radar option in the web-interface.

But.. the FLARM option is not working... (I've enabled the option on the setting page).
 
And if I test the latest release of PepperJo 'stratux-v1.5b2-flarm-fe12ed3.zip', FLARM is displayed in the Traffic list (green).

Maybe I did something wrong?

So I could use PepperJo's image but I like the other options you give.

Hope you can help me out with this problem :-)

VirusPilot, 2019-05-19:
I have the 'Stratux Europe Edition 1.5b2-eu002' running and could receive FLARM targets. Are you sure that you had a valid GPS fix?

CHRE2, 2019-05-19:
Yes. GPS location is found.
 sudo nano /root/stratux/ogn/rtlsdr-ogn-0.2.6/stratux.conf also show the coordinates.

I'll try an other antenna, maybe that will do.

VirusPilot, 2019-05-19:
Just committed my latest GPS changes to my fork, feel free to try it out.

CHRE2, 2019-05-19:
Used my roof antenna that usually is connected to a flightaware pro stick + for 1090 adsb, now I got 1 FLARM station!

b3nn0, 2019-05-20:
So this can be closed?
Note: You can also check the new Flarm Status page to check if the decoder is running and receiving correctly, and make sure it is connected to localhost as an APRS server. When that is the case, everything should be working fine.

CHRE2, 2019-05-20:
Jup. Can be closed now. It takes a while to start it up I guess, but after a several minutes it is running (the page on the FLARM status page is shown).
Just ordered a 868 MHz antenna on Amazone, will check it out :-)

closed 2019-05-21 by b3nn0