stratux/stratux #10: Switching to a Raspian Stretch based image

b3nn0, 2019-06-15:
The last couple of versions were based on an original Stratux image.
I.e. a script fetched the latest Stratux image, checked out the European version from git, installed it and that's it.
However, the base of the original Stratux image (Raspbian Jessie) is quite ancient by now.

I've done some tests and see no reason why we should stick with such an old version. I've created a Stratux Image that is based on the most recent Raspbian Stretch image and it works quite well.

Benefits:
- more control of what is on the image
- Raspberry PI 3b+ support (not supported by Jessie)
- better future-proofing
- Possibly better drivers and firmware

Risks:
- Maybe some consequences I don't see yet? In my tests, it all looked good.
- Limited upgradability: The newer Stratux upgrade files (.sh scripts) will still work on the older images. However, they can of course not upgrade the Raspbian base image, just the Stratux software. This means there will be more variance in the software that people are running. Some will have done a clean install of the new Raspbian base, some will be on an old Jessie Stratux that has been upgraded to the latest Stratux code.
- Possibly more testing of update files needed against both base system versions (or deprecate the old one as "unsupported but probably working")

Some things need to be evaluated: https://github.com/b3nn0/stratux/wiki/RaspberryPI-3-b---Testing (anyone using a Stratux with OLED screen?)

My opinion: either go for it, or wait for the new Buster release and go with that (Debian Buster will likely be released in July, Raspbian will likely follow 2-3 months later, I would guess).

What are your opinions about this?

russgreen, 2019-06-21:
Do you have any idea if and when the main Stratux image will use the new Rasbian?  I would suggest waiting for the new version after July.

b3nn0, 2019-06-21:
I do not. The issue in the main Stratux github where this seems to be worked on is this:
https://github.com/cyoung/stratux/issues/768
but there doesn't seem to be that much activity.

b3nn0, 2019-06-24:
Raspbian Buster was released today with the Raspberry Pi 4 announcement.
I have created a branch where progress can be tracked. Further discussion here:
https://github.com/b3nn0/stratux/pull/14

closed 2019-06-24 by b3nn0