stratux/stratux #19: FLARM-NMEA altitude doesn't fit for FLARM targets

maehtricks, 2019-07-14:
1. Stratux version: 1.5b2-eu008 BETA
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: u-blox 8

    AHRS
      - [X] yes
      - [ ] no

    power source: -

    usb cable: -

3. EFB app and version: SkyDemon
    EFB platform: Android 9
    EFB hardware: Nokia 6

4. Description of your issue:
Traffic relative altitudes are ok if I'm using GDL90 and MSL altitude in stratux settings. If I'm using FLARM with AirConnect in SkyDemon, FLARM targets are shown as as too high (today +300ft @ QNH 1018). ADS-B targets are fine. Looks like a sign error in FLARM-NMEA code to me.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

[image]
[image]
[image]
[image]
[image]

VirusPilot, 2019-07-15:
We had that issue earlier with GDL90, that is why the MSL switch was introduced, it should be in OFF position for SD, like the attached picture, therefore I am confused about your statement "Traffic relative altitudes are ok if I'm **using GDL90 and MSL altitude** in stratux settings". Traffic should be ok when using **GDL90 and HAE** right? The fact that also AirConnect over FLARM-NMEA also shows a difference in altitude leads me to the conclusion that the FLARM glider on your screen may not have sent a correct altitude.

[image]

b3nn0, 2019-07-15:
I have just looked into the issue and you seem to be correct. From what I can see, the baro altitude is compared to the gliders GPS altitude. I think I already have a valid fix, will test it later today and report back.
@VirusPilot the switch is not really applicable here. Flarm-NMEA requires that the _relative_ altitude is provided in the PFLAA string, not the absolute altitude. Therefore "-300ft" is just "300 ft below", no matter if the rest of the system is calculating in pressure, ellipsoid or MSL altitude.

VirusPilot, 2019-07-15:
@b3nn0 I agree, but @maehtricks seemed to have used this switch for his GDL90 connected stratux to "correct" the obvious discrepancy.

b3nn0, 2019-07-15:
Just committet the likely fix: https://github.com/b3nn0/stratux/commit/7a10a80f3c062f2ec06ade287155c424dc79132b
Feel free to test and report back if you know how.

maehtricks, 2019-07-15:
Sorry, description was little bit confusion... mixed up MSL and HAE. But I meant what you're seeing in the screenshots :) FLARMS were at my level, even different ones.

I'm happy that you found the issue, keep up that good work! I'll build and verify it later on, have a FLARM device next to me.

maehtricks, 2019-07-15:
Looks better now, but traffic is still shown 100ft too high in SD using NMEA. There must be a small difference in the calculation. Do we have to care for geoid sep over here probably?

b3nn0, 2019-07-15:
Hmm, that is strange. Well, one difference is, that in GDL90, Stratux just transmits the device altitude and the traffic altitude. In Flarm, the computatin of relative altitude is done by the Stratux and SkyDemon only receives "Traffic in +-xxxft". So the computation is done entirely on the Stratux side.

With respect to geoid/ellipsoid alt: You might be right. According to some posts in the OpenGliderNet google group, they are reporting geoid altitude, but only if the geoid separation is configured in the ogn config file. The stratux ogn config file sets this to 0, so I assume that the OGN stack reports ellipsoid altitude in this case.

I have done another commit that will do the calculation based on ellipsoid altitude for flarm targets. Sadly, I don't have a Flarm device at home to test. Can you try again with this and tell me if it worked out?

maehtricks, 2019-07-15:
It matches up perfectly now. Thanks, I think we can close this issue :)

There's probably the same thing for web radar. Traffic is shown/announced as too high (+300ft with QNH1018).

closed 2019-07-16 by b3nn0