stratux/stratux #20: Improved serial support

b3nn0, 2019-07-16:
I received some questions over time about serial output.
Stratux is already able to dump GDL90 via Serial. However, it only supports the CP2102 USB dongle provided by Stratux, or if you want to use a custom CP2102, you need to change its serial to work with Stratux.

It might be of interest to
- Support more different dongles. From a code perspective, it wouldn't be any different to handle an FTDI or similar (or even the Raspi's serial pins if 3.3v is enough and the AHRS module isn't used)
- Instead of dumping GDL90 via Serial, we might want to make it possible to dump Flarm-NMEA aswell, as some avionics seem to prefer that (which?)

Implementing the first is fairly simple. Add a few udev rules for different dongles to make them available under a well-known location, then open that as our serial port (only one serial supported for the beginning.

Implementing the second is trivial - add a settings switch and a simple if-condition in the serial code to dump Flarm-NMEA instead of GDL90.

Any opinions on this? If you would like to see this, let me know. And also let me know why you want/need this so I get a feeling for the priority. Do you need support for different baud rates?

Spoiler: With a quick try, I was able to easily receive Flarm-NMEA via minicom on a second RasPi, using an FTDI set to 3.3v on the transmitting side and the on-board pins on the receiving side - but it's not userfriendly as of yet. GPS was disconnected here, so coordinates don't show.
[image]

VirusPilot, 2019-07-17:
One application could be to feed a non ADS-B out transponder with GPS, a further RS232 converter would be required in addition .

maehtricks, 2019-07-17:
Another use case could be feeding Openvario with data.

SarahAnderson21, 2019-09-08:
This keeps coming up, and yes, it would be a valuable addition.  IF the FLARM NMEA is enough to satisfy numerous commercial FLARM traffic displays, displaying only ADSB traffic. These are all wired serial devices, usually fed by powerflarm or (EU) original flarm tx/rx hardware.

Commercial US "powerflarm" does not receive UAT which is a huge drawback. The only way to use robust in panel (existing) traffic displays in gliders is to homebrew it. As far as I know..

What happened to the efforts in 2016/2017?   https://github.com/cyoung/stratux/issues/571

scsirob, 2020-02-29:
The stratux.conf file in EU13 shows a line "SerialOutputs": null,
Is this functional at all? If so, what is the syntax for this line? We're interested in feeding FLARM NMEA into an AvMap V.

b3nn0, 2020-03-01:
This has never been implemented. The SerialOutputs config variable is for GDL90-out-via-serial that is also found in upstream Stratux. It will be configured automatically if a suitable serial-out device is connected.

I'm currently not planning to implement this, pull requests are welcome.

vontresc, 2020-04-07:
So I got bored the other day, and hooked up this branch of stratux to my oudie.

I used a prolific usb to serial converter since it was set up to work directly with the USB/Serial port on the oudie. As a result i had to play around with the udev detection a bit.

https://github.com/vontresc/stratux

[image]
[image]

b3nn0, 2020-06-05:
This will be implemented in the next version, see here:
https://github.com/b3nn0/stratux/wiki/Stratux-Serial-output-for-EFIS's-that-support-GDL90-or-Flarm-NMEA-over-serial

closed 2020-06-05 by b3nn0

vontresc, 2020-06-09:
Thanks for getting this in the standard build. Tested it last night, and it worked well. That’s the traffic warning from my friend’s RV-6 with UAT out right in front of me.