stratux/stratux #21: Better support for SoftRF / Transmitting Flarm

b3nn0, 2019-07-26:
Hi,
Stratux has rudimentary support for SoftRF. This is implemented by
- starting a dump1090 process with --net-only
- reading SoftrRF dump1090 output via Serial, forwarding it to dump19090
- reading traffic from dump1090

This has a couple of drawbacks:
- it interferes with the dump1090 process for the SDRs, so you can only have EITHER a SoftRF connection OR a 1090ES SDR enabled
- We have two GPS in the device, the SoftRF one is ignored for Stratux

Is it worth implementing better Support for SoftRF (especially the T-Beam)?

Benefit:
- We can easily transmit Flarm
- We can also receive OGNTP and PilotAware if desired
- We could use the SoftRF builtin GPS for Stratux
- The second SDR would be free for UAT reception

Drawbacks:
- SoftRFs Flarm reception range seems to be quite a bit worse than the OGN stack with an SDR
- Another PCB - different case is needed (could provide a 3D printable model that fits a Stratux+SoftRF build)

Implementation:
I think it would make most sense to parse the SoftRF NMEA stream instead of the dump1090 stream, as it provides more information.

b3nn0, 2019-08-21:
Closed due to lack of interest.

closed 2019-08-21 by b3nn0

Michal2184, 2020-06-07:
Hi
With new update I decided to buy TTGO T-beam board and BMP280 as would like to try it out.   Don't think many people has done such an upgrade so guide is no were to be found. Can you shed some light how to get this working together?  I presume t-beam need to be loaded with OGN tracker, than pairing it with BMP280? . Will it just work after plugin into stratux usb.....well i think it need a bit of tweaking? 

Any sugesstions will be great! Should have t-beam this afternoon :)