stratux/stratux #22: Encrease number of devices excepted from traffic list

Koudy232, 2019-08-07:
For tracking and anti-collision purposes I am having on board FLARM/OGN device together with Stratux and Mode S transponder. I realized, I can file FLARM/OGN hex code on the field form mode s transponder and it works the same way (particular device is not send as nearby traffic). Would be possible in the future release  increase number of devices instead of one?

b3nn0, 2019-08-08:
Already thought about this myself, not sure when I will get around to implement it - but it does make sense IMO.

b3nn0, 2019-08-21:
This has now been implemented. You can set multiple exclusions seperated by Comma (AAAAAA,BBBBBB,CCCCCC).
Note: I recommending setting your Transmoder Mode-S as the first entry in that line, as this is the one that will be reported as the Ownship Mode-S via GDL90 (not entirely neccessary though).

Some things that might be a problem with this:
Right now, whenever we receive a signal (via Flarm or ADS-B) which matches our ownship mode-S, we assume that this is the most recent signal for us - and use Information from that signal as our source for pressure altitude, latitude, longitude, track, ...).
If this information is different for the two devices (e.g. Flarm has bad GPS reception and reports altitude as 300ft too high), we might see some jumps in the GDL90 stream, reporting sometimes higher, sometimes lower.
Will have to see if this is only a theoretical problem or if it can cause problems in practice.

closed 2019-08-21 by b3nn0