stratux/stratux #23: Foreflight AHRS

russgreen, 2019-08-21:
1. Stratux version: 1.5b2-eu010

Foreflight is not displaying attitude information.

b3nn0, 2019-08-23:
AHRS works just fine here in Foreflight and SkyMap.
Please check if
- the AHRS in the WebInterface is working
- you have a GOOD GPS signal
AHRS data is not sent to the EFB when you have no or bad GPS reception (needs to be <30m iirc).

russgreen, 2019-08-24:
I have really good GPS accuracy and the AHRS wors in the web interface.  Just Foreflight shows "No attitude information".

Fujiflyer, 2019-10-15:
I also have perfect GPS, AHRS works fine in the Web Interface, but no Attitude in ForeFlight!
Traffic and GPS works fine in ForeFlight!

b3nn0, 2019-10-15:
I'm afraid I can't really help you, since I don't have Foreflight. Maybe @VirusPilot can say something? But last I heard from him, he said that it was working just fine for him...

VirusPilot, 2019-10-15:
Yes, confirmed. It just worked right away. Unfortunately my Foreflight license expired so I can't do any further test.

Fujiflyer, 2019-10-15:
Hm???  any Idea what i should try?
Stratux version: 1.5b2-eu010

b3nn0, 2019-10-16:
Sadly, not really. It works just fine in Sky-Map aswell.
The only thing I can think of would be
- re-imaging the micro sd
- check if the same happens with upstream Stratux (cyoung/stratux)
-> Since the AHRS code is unchanged from upstream, I would assume that it behaves the same aswell.

Fujiflyer, 2019-10-18:
Hey, thanks! With new Version 1.6 eu everthing seams to work !!!

b3nn0, 2019-10-18:
That's interesting, because nothing changed in that respect at all.
So I assume that basically the re-imaging of the micro sd did the job. Closing this for now.

closed 2019-10-18 by b3nn0