stratux/stratux #24: Stratux as a position logger

Koudy232, 2019-10-18:
From my point of view very good feature would be position logging (together with G-load and AHRS data). I would suggest .IGC or .KLM format.  I know the basic of codding, so I can write this part but probably not be able to implement it to the Stratux core code, would anybody help?

b3nn0, 2019-10-18:
Out of curiosity: what is your motivation for such a future?
Aren't the EFBs much better at doing this? E.g. SkyDemon does nice logs that can be viewed on an aviation-centered map with airspaces, etc, and even exported as KML or GPX.
When would you prefer to use a Stratux based logging? When flying completely without an EFB or something?
Especially, since KML doesn't really have support for AHRS data (or does it?) so we could only add that as some additional numeric attributes that nobody really understands. The G-Load would potentially make sense there, I agree.

From a first glance, IGC doesn't support AHRS either (or does it?)

Koudy232, 2019-10-18:
Thank you for quick replay. Here i my motivation, I am managing a few microlights. I would like to have logs of all the flights without using any EFB. Standard loggers (for example gliding IGCs) usually save only GPS position, but big plus would be if AHRS and G load data would be save as well (It happened to me this year that somebody exceeded allowed G-load and damaged wing spar). I did the research and you are right that neither IGC nor KML support AHRS and G-load data. Solution would be save all those data in CVS and use special software for analyzing - for example CloudAhoy which is able to interpret it: http://help.cloudahoy.com/cloudahoy-csv-file-format/.

b3nn0, 2019-10-19:
Alright, that does sound like a reasonable motivation and I do see your point.
As for implementing this:
I would say it is impractical to write files like this during flight, since it is quite common that the Stratux loses power. So we wouldn't know when to e.g. close our KML files (</Placemark></kml> or something). For logging, it is probably more practical to use a format that handls power-loss well.
I.e. logging should probably be done to SQLite. This also gives us the option to possibly log more stuff in the future (traffic, weather, ...)
We would then, of course, need some conversion functions that transform this SQLite log into KML, CSV or whatever. But that doesn't seem too hard to do.
The more annoying part - at least for me, since I'm more of a backend developer than a  web developer - is the UI that lets the user configure logging and export tracks in a desired format.
Are you familiar with bootstrap web development? That is certainly something where help would be appreciated.
My recommendation:
A new page that cont ains everything we need a configuration section
- Log ownship information [checkbox]
- (in the future possible also log traffic, log WX, ...)
- Log retention days [numeric input]
[Display a rough estimate for the log file size]

Additionally, we need an export section
- From [datetime]
- To [datetime]
- Format [drop down selection]
- Download [button]

Optionally
- have a table on the page which contains the individual flights in the database and lets you download them directly in the desired format.

Koudy232, 2019-10-19:
Your solution seems very elegant. I have no experience with bootstrap. Give me some time I will look at it - I would like to contribute implementing this feature.

Koudy232, 2019-12-30:
I looked at it, but unfortunately to do this is out of my programming skills. Could any body else help? 
Maybe already created Logs page can be use for this purpose.  What do you think?

closed 2020-12-09 by b3nn0