stratux/stratux #25: Other connectability options than Wifi hotspot

b3nn0, 2019-10-19:
Hi,
again, I'm looking for opinions:
Personally, I do not like the fact that I have to connect to the Stratux Wifi hotspot for the system to work.
Reasons:
- Android devices will not be able to connect to the internet via mobile network simultaneously
- iOS devices can, but it requires changing the DHCP configuration on the Stratux via SSH
- The Stratux, in practice, never has an internet connection (except if people take it home and connect it via ethernet)

Motivation for other connection options:
- EFB devices can keep their internet connection via mobile network
- the stratux could potentially use that same internet connection during flight, to potentially do a ton of nice things:
- We could connect to an Mlat service (e.g. adsbexchange or something) and fetch mlat traffic from there and publish it via GDL90/Flarm-NMEA
- We could potentially fetch weather over the internet and therefore simulate UAT-WX
- We could potentially enable easier online updates or update notifications

Possible implementations:
- Wifi tethering
the Stratux connects to it as a client. I personally already do it this way, but it's not a simple configuration. Getting the implementation right is probably difficult.
- Bluetooth connection:
The Pi3 already has bluetooth on-board. However, most EFBs will not accept GDL90 via BT directly. Could we convince EFB manufacturers to implement this?
- Bluetooth Tethering
That should probably work the best. We still use sockets for communication, so the EFB should be easily compatible. So my vote would be for this, if it is possible.

Challenges:
- BT tethering is probably not exactly straight forward, especially with respect to pairing, etc. It will require significant work.
- Additionally, i'm not sure if it is possible to set the IP Address for the Stratux to 192.168.1.1, as some EFBs require it for Flarm-NMEA. GDL90 should work just fine.
- Since we are not the DHCP provider any more, we need to have a different mechanism for discovering the clients that should receive GDL90. This should be doable by reading the ARP table - that's probably enough.

colmoschin94, 2019-10-20:
Hi b3nn0, for what is worth I've modified my Stratux Image (based on your project) to make it work as a wifi client, the only method that seems to work always reliably is this one: https://imgur.com/a/wJa5J
Operating the Stratux as a Client gives (of course in my opinion) more flexibility than the standard AP mode, expecially because it allows you to use an external AP or phone as main connectivity device (which might provide data to EFB, Stratux, etc). I've tested it in the past few weeks with SkyDeamon and I've found it performs much better (expecially when it comes to download files during flight, eg. plates or weather data), the only issues so far (as you mentioned before) are that user must set static IP to send gdl90 data and the lack of a webui to switch between AP/Client mode.

Thanks a lot for your amazing work!

Matt

b3nn0, 2019-10-20:
Yes, I'm well aware of that option and using something similar (in fact, I have a secondary Wifi stick which is configured in client mode, so I can choose between both modes on the fly).
Solving the issue of manually entering the IP is not too hard - the kernel's ARP table has the IP and we can read it from there.
One main problem I see is, that many EFBs expect Flarm-NMEA at 192.168.1.1, which is not true any more in a situation like that.
And of course the fact that implementing this in Stratux is quite a bit of work.

colmoschin94, 2019-10-20:
Sure I understand!
Thanks a lot for the feedback.

Matt

From: Adrian Batzill
Sent: Sunday, October 20, 2019 5:35 PM
To: b3nn0/stratux
Cc: colmoschin94; Comment
Subject: Re: [b3nn0/stratux] Other connection options than Wifi hotspot (#25)

Yes, I'm well aware of that option and using something similar (in fact, I have a secondary Wifi stick which is configured in client mode, so I can choose between both modes on the fly).
Solving the issue of manually entering the IP is not too hard - the kernel's ARP table has the IP and we can read it from there.
One main problem I see is, that many EFBs expect Flarm-NMEA at 192.168.1.1, which is not true any more in a situation like that.
And of course the fact that implementing this in Stratux is quite a bit of work.

spreeni151, 2019-11-15:
>     * Android devices will not be able to connect to the internet via mobile network simultaneously

This is not completely true! You can try the software Speedify (www.speedify.com) which uses a VPN connection for transmitting at both ways, wifi AND mobile network. Speedify is free to use unless you do not produce traffic more than 2GB per month. This could be a more easy way to use Stratux and internet at the same time on Android Devices.

b3nn0, 2020-01-04:
Now found a different solution (and implemented it):
Next version will have experimental "WiFi Direct" support.
Result will be, that you can set a WiFi Direct WPS Pin and use Wifi-Direct to connect, which should work on most modern Android device).

Advantages:
- Android doesn't screw up the routing and keep mobile data alive
- There is still a normal hotspot to connect to if the device doesn't support WiFi-Direct (but it will be prefixed with "DIRECT-", so e.g. DIRECT-stratux

Disadvantages:
- Android doesn't remember the connection as with WiFi. You need to manually connect and enter the PIN every time

Result will look like this:
[image]

closed 2020-01-04 by b3nn0