stratux/stratux #27: SD-Card is not booting

maggo1404, 2019-11-10:
1. Stratux version: v1.6r1-eu011
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: UBlox NEO-M8N

    AHRS
      - [x] yes
      - [ ] no

    power source: 2A-Powerbank

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Skydemon

    EFB platform: (e.g., iOS 9.2) iOS

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 4 Cell

4. Description of your issue:

The Raspi kann not booting with this Image Version. I sea only a rainbow image on the monitor( conected via HDMI-Interface). The standard image vom cyoung is booting and creat a WIFI-Network. The Raspi is a 3B Model

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2019-11-12:
The image is working just fine for me and many other users. Not really sure what the issue is in your case.
The base image is just the same as can be found on the raspberry pi website - with some modifications applied. The rainbow screen indicates, that the kernel image can't even be found correctly.
Maybe try re-downloading and re-writing the SD card? It should work just fine...

maggo1404, 2019-11-15:
I have found the Problem. The SD card is demange. I have change the SD card and now the system is running.

closed 2019-11-15 by maggo1404