stratux/stratux #28: featue: no ip adresse change required for skydemon

maggo1404, 2019-11-23:
- Set the a secend ip adress to the wlan0 interface (192.168.1.1/24 command "ifconfig wlan0:1 192.168.1.1/24"). i have add the command to the script /etc/rc.local
- change the dhcp server settings
the answer with the network setting:
ipadresse 192.168.10.X with network mask 255.255.0.0

you can use the stratux with foreflight (GDL90) and skydemon (FLARM with AIRCONNECT) with out change the setting.

b3nn0, 2019-12-29:
Sounds like a good idea - might add that at some point.

b3nn0, 2020-07-21:
I've decided not to implement this.
While it would make it slightly easier to work with SkyDemon, it has the disadvantage that Stratux will claim the whole subnet, which makes it impossible to connect it to your home network if it is also in the 192.168.*.* range - and also will break some configurations with secondary Wifi dongles in client mode.

closed 2020-07-21 by b3nn0