stratux/stratux #31: X-Plane interface

neckmann, 2020-01-15:
The beta implementation of the X-Plane interface for Garmin Pilot worked quite well for me.

Can you implement the interface in the normal version?

b3nn0, 2020-01-16:
Are you sure that works properly? When I tested it a few months ago, it didn't work too well. Traffic was not displayed at all with Garmin Pilot for Android.

neckmann, 2020-01-17:
[image]
[image]
[image]
[image]
[image]
[image]

neckmann, 2020-01-17:
For me it is going well.
The only mistake is a flickering traffic (see appendix), but still better than no traffic. 

[RPReplay_Final1579292428.zip](https://github.com/b3nn0/stratux/files/4078931/RPReplay_Final1579292428.zip)

Maybe the XPlane output can be implemented as an experimental function?

b3nn0, 2020-01-17:
Alright, fair enough, added it again for the next version.

closed 2020-01-17 by b3nn0

norbeta, 2020-04-23:
Just cloned your repo and installed it and can confirm that the X-Plane interface is working correctly. 
When installing it, it's important to either wipe the config or change capability to 18 for port 49003.

tmmtaha75, 2020-06-19:
I would also suggest extending the X-Plane capability to Foreflight. Foreflight is notoriously known for disconnecting the app from the internet when connected to any ADSB device because the majority of its users are in the U.S. and can receive weather, NOTAMs, etc via the device itself. In most of Europe, we don't have UAT towers and can't receive any of this vital info in flight; our only hope is sporadic internet at low altitudes to update the weather info. Foreflight in X-Plane mode stays connected to the internet; we will be able to receive traffic and AHRS data from STRATUX and the rest from the internet.