stratux/stratux #33: Trig TN72 through FTDI not detected

scsirob, 2020-01-29:
1. Stratux version:
stratux-v1.6r1-eu013
2. Stratux config:

    SDR
      - [ ] single
      - [ X] dual

    GPS
      - [ X] yes
      - [ ] no
      type: TRIG TN72 via FTDI serial<->USB

    AHRS
      - [ ] yes
      - [x] no

    power source: 2.1A 5V Samsung charger

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
EasyVFR 4.0
    EFB platform: (e.g., iOS 9.2)
Windows 10
    EFB hardware: (e.g., iPad Mini 2)
Desktop PC

4. Description of your issue:
TRIG TN72 GPS not detected. Status window tells me "Serial port (Not communicating)"
Linux kernel shows FTDI converter recognized and connected as ttyUSB0

Jan 29 12:17:07 raspberrypi kernel: [  162.530574] usb 1-1.2: Product: AvMap USB Serial Converter
Jan 29 12:17:07 raspberrypi kernel: [  162.530586] usb 1-1.2: Manufacturer: FTDI
Jan 29 12:17:07 raspberrypi kernel: [  162.530597] usb 1-1.2: SerialNumber: A7VHM9ZE
Jan 29 12:17:07 raspberrypi kernel: [  162.542374] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
Jan 29 12:17:07 raspberrypi kernel: [  162.542634] usb 1-1.2: Detected FT232RL
Jan 29 12:17:07 raspberrypi kernel: [  162.547323] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0

NMEA output shows when opening ttyUSB0:

pi@raspberrypi:~ $ cat /dev/ttyUSB0
$GPRMC,,V,,,,,,,,,,N,V*29
$RAIM,000000,000000,000000,000000,000000,000000,000000,000000,111,0,00000000*3A
$GPGGA,,,,,,0,00,,,M,,M,,*66
$ACVT,0,,,,,,*1C
$GPVTG,,T,,M,,N,,K,N*2C

b3nn0, 2020-01-29:
This is more or less intended, as FTDIs can be useful for anything - GPS In, NMEA Out, GDL90 out.
To make this work, you have to make the GPS available under /dev/ttyAMA0 or /dev/flarm with an appropriate udev rule in /etc/udev/rules.d/10-stratux.conf. These will both be considered as Serial NMEA IN.

scsirob, 2020-01-29:
Thank you, I'll play with those and see what comes up.

closed 2020-01-29 by scsirob

scsirob, 2020-01-29:
The method works, but not entirely as I had hoped it would. 
It turns out the /dev/ttyAMA0 and /dev/flarm links assume flarm baudrates of 38400 Baud. Using stty I could set baud to the Trig 9600 for a short while but I guess the Flarm code overwrites the speed. So I added the FTDI to generate a symlink to /dev/prolific%n like the PL2303 and after setting speed to 9600 a couple of times, NMEA was detected.