stratux/stratux #37: No traffic in Foreflight

jankoenraad, 2020-03-08:
1. Stratux version: 1.6r1-eu013 (45f654956e)
2. Stratux config:

    SDR
     - [ ] single
     - [ ✓] dual

    GPS
      - [✓ ] yes
      - [ ] no
      type:vk162

    AHRS
      -  ✓ yes
      - [ ] no

    power source: external battery

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Foreflight 12.1.1 and Skydemon 
    EFB platform: (e.g., iOS 9.2)
iOS 13.3.1
    EFB hardware: (e.g., iPad Mini 2)
iPad air (3rd generation)
4. Description of your issue:
Traffic is shown in Skydemon, but at the same time the device is not found in Foreflight and therefore no traffic is shown. The original Stratux version provides traffic to foreflight without problem.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-03-10:
There are a lot of people using Stratux EU with Foreflight without issues.
Additionally, the code that feeds your EFB is practically the same as upstream Stratux, so I can only assume user error.
Maybe Skydemon is still running in the background and steals the data or something? Try rebooting both, Stratux and the iPad and try again.

maggo1404, 2020-03-10:
Try, the ipadresse from stratux change to 192.168.1.1 and the ipad wifi interface is set to dhcp.

jankoenraad, 2020-03-20:
I was able to reproduce the issue

1. connect with stratux wifi network
2. open Foreflight
3. traffic is shown in foreflight as normal
4. open Skydemon simultaneously
5. "Go flying" in skydemon and select "GLD90 compatible device"
6. Go back to Foreflight. Traffic is not shown. 

I can imagine that this is not a Stratux specific issue.

b3nn0, 2020-03-20:
Yes, this can be expected. Only one application can listen on UDP:4000, where the traffic is sent, at the same time.
I assume in your case, that Foreflight stopped listening, so SkyDemon took over and didn't let go when switching back to Foreflight.
Basically, this protocol is not suited for a case where multiple applications on the same device should receive the data.
Closed.

closed 2020-03-20 by b3nn0