stratux/stratux #38: Europe Edition 1.6r1-eu013 with SoftRF USB Dongle

tmmtaha75, 2020-04-18:
1. Stratux version: https://www.amazon.com/gp/product/B01LZTZ06C/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
I replaced the low power SDR with the older version to work with the FLARM build
2. Stratux config: Stratux Europe Edition 1.6r1-eu013

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: internal GPYes WAAS GPS receiver

    AHRS
      - [x] yes
      - [ ] no

    power source: External battery

    usb cable: High load

3. EFB app and version: Foreflight latest version 12.2.1 (e.g., WingX Pro7 8.6.2)

    EFB platform: iOS 13.4.1 (e.g., iOS 9.2)

    EFB hardware: iPad Pro 10.7 inch (e.g., iPad Mini 2)

4. Description of your issue:
I connected a preconfigured SoftRF USB dongle with GPS and FLARM NMEA in. I can see OGN traffic in the Stratux software radar but not in Foreflight. Is there a way to make OGN traffic visible in Foreflight? Thank you so much for your great contributions to the European GA community!
 
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-04-19:
Hmm, I don't have Foreflight, but the Flarm-NMEA-in traffic is handled just as any other traffic (e.g. ADS-B). Stratux does not differentiate between those once it is received. I can see the traffic just fine in SkyDemon.
Are you sure this is not related to some filter option in Foreflight?

tmmtaha75, 2020-04-19:
Thank you for your quick reply. I will check with Skydemon, but do I need to buy the FLARM decoding subscription? I think if I get to see OGN in EFBs in Europe along with FLARM with the SoftRF, ADS-B  mode S proximity and free up the UAT SDR for the UK, it would be the best solution in Europe for now. By the way, I'm an IRI in Hungary and it would be great if we could communicate on the side. My email address is tmmtaha75@gmail.com.

b3nn0, 2020-04-19:
Your SoftRF does the decoding, so no, no seperate license is required. It should work just fine with both, SD and Foreflight, as long as the EFB does no filtering for some reason and there is traffic around you.
In SkyDemon, select the GDL90 Protocol.

VirusPilot, 2020-04-19:
Maybe you should test first with FF without SoftRF but with a "standard" GPS (like the GPYes v2) first. How was your SoftRF GPS performance?

tmmtaha75, 2020-04-19:
Thank you! Everything works as expected with and without the SoftRF. The only difference is that I couldn't see the "OGN" traffic in Foreflight while everything else was fine.

tmmtaha75, 2020-05-19:
Dear all: Thank you for your help thus far. I have 2 SDRs: UAT and ADSB. I also have a SoftRF GPS dongle that receives OGN and FLARM connected to my Stratux. I don't have an internal GPS. In Stratux Europe Edition 1.6r1-eu015 when I deactivate UAT in settings I am able to see ADSB and OGN traffic, but when I activate UAT I can't see OGN traffic even though I should because the SoftRF dongle is connected and sending the GPS and OGN to the Stratux and should be receiving FLARM and OGN separately from the UAT SDR. Any suggestions/comments?

b3nn0, 2020-05-19:
From what I know, the SoftRF dongle does not receive FLARM and OGN simultaneously. You have to configure it and will receive _either_ flarm, _or_ ogn. Not both at the same time.
If you disable UAT, it will work, because then the UAT SDR will be used with the OGN stack to receive Flarm+OGN (in addition to the SoftRF dongle).

closed 2020-06-06 by tmmtaha75