stratux/stratux #40: AirNav Pro attitude problem with Stratux EU edition

killahill, 2020-04-30:
Hi!
I can't get AirNAV Pro ATTITUDE to work with Stratux EU edition. Works perfect with non EU Stratux.
Since version 1.4 when upload image with flarm capability can't use AHRS data.

killahill, 2020-04-30:
I just found that with non EU version stratux-v1.6r1-377d22b403 doesn't work too.

VirusPilot, 2020-04-30:
Did you try this configuration?

[https://github.com/0sander/stratux_air_nav_pro](url)

killahill, 2020-04-30:
Yes! In fact AirNavigation Pro software is connected to Stratux, Traffic is displayed, but AHRS data seems not to be valid.

b3nn0, 2020-05-01:
Stratux EU doesn't handle AHRS any different.
Am I correct to assume, that you are using the GDL90 protocol for communication, Not Flarm-NMEA?
The Flarm-NMEA protocol doesn't support AHRS.
So even though Stratux can output Flarm-NMEA, you are still following the previously posted guide?

Additionally does AirNavPro support AHRS for the X-Plane protocol?
If so, you might want to try that. Stratux does output attitude data via X-Plane protocol, but I've never tested it myself.
Note that you have to re-flash the MicroSD to enable X-Plane output (or configure the output manually in stratux.conf if you know how).

killahill, 2020-05-01:
yes pal
it is GDL90. I am receiving traffic, but no AHRS.  Stratux 1.4r5 /NON EU/ works perfect with AHRS. Yesterday tried 1.6r1 /NON EU/. Doesn't work too. Version 1.5r2 /NON EU/ doesn't work too. Tried X-Plane protocol, but didn't connect. I am nut sure is the IP 192.168.1.1 correct one.

b3nn0, 2020-05-01:
Usually it's 192.168.10.1 in default configuration (can be changed in settings).
ANP is known to be fairly wonky when it comes to external device support. As the same happens randomly with Non-EU Version, it is not a Stratux-EU issue.
Most likely, it's either a configuration issue or a bug in ANP.
The AHRS Code in Stratux hasn't changed at all since 1.4, and hasn't changed in the EU version either.

killahill, 2020-05-01:
Yes may be is not Stratux-EU issue. I have some memories that lost AHRS when added flarm support package on version 1.4. /not with image/. It is really strange. With clean Stratux 1.4.r5 works. Will try to find out way to connect as X-Plane. Thanks a lot!

killahill, 2020-06-15:
Yesterday updated AirNavigation PRO. The issue with AHRS from Stratux is solved. Regards.

closed 2020-06-15 by b3nn0