stratux/stratux #42: Bad World Map image

Ulli71, 2020-05-05:
The file /web/img/world.png fills the mediterranean sea and the baltic sea with the 'land' - color so it appears as a land mass. Accordingly it shows up in the GPS view.

b3nn0, 2020-05-05:
Feel free to provide a better image.

Ulli71, 2020-05-06:
Hi B3nn0,
I hope this reply works.
I edited the existing image. I dropped blue color on the red sea,
mediterranean sea, the baltic sea, the black sea and other seas in the
former soviet union. As I edited the existing image the dimensions and
proportions are not touched.
See attachments.
Regards
Olaf.

b3nn0, 2020-05-06:
Sorry, E-Mail Attachments don't work with github notification emails.
I think you should be able to add the file when you reply in this issue on github (I hope it won't be scaled down. If it is, you will have to upload it somewhere)

Ulli71, 2020-05-06:
I am sad to read that my first contribution didn't work. Here my 2nd attempt. Sorry folks, I am learning..
[image]
[image]

b3nn0, 2020-05-06:
Perfect thank you. Will be in the next version.
(Just for the record, usually contributions are done with pull requests. I.e. you need to fork the repository, clone it to your machine, do the modifications, commit, push and pull request. But for something simple like this, just posting the image here is of course just fine).

commit da37dc0a (2020-05-06): update worldmap with fixed water areas, thanks to Ulli71, fixes #42
closed 2020-05-06 by b3nn0