stratux/stratux #43: Option to disable GDL90 or FLARM output

Krasnall, 2020-05-12:
At least one EFB (EasyVFR4) is able to receive traffic information from GDL90 and FLARM protocols at the same time. For bearingless targets, it creates doubled targets. It also shows dummy aircraft, even if the EFB is able to receive FLARM messages - which as I understand from previous discussions are better than GDL90. 

It would nice if Stratux had a switch to limit output protocol (GDL90 or FLARM) at least for traffic information.

VirusPilot, 2020-05-12:
While waiting for b3nn0's reply, can I ask you which GDL90 Altitude switch setting you are using for Stratux Europe with EasyVFR? The GDL90 spec requires HAE (height above ellipsoid) and therefore some EFBs only process altitude information properly from Stratux if the switch is disabled (e.g. SkyDemon). However there are other EFBs that expect MSL instead, although this is not according to the spec. We want the Wiki to cover all relevant EFBs and the recommended settings but we don't know what EasyVFR is expecting. Can you help to find this out?

Krasnall, 2020-05-12:
I will try to find out which altitude is expected by EV4.

In the meantime, I found out it's possible to simply disable the GDL90 output by removing it from stratux.conf file. So far I did one test on the ground and it seems that it's behaving as expected - I got a single traffic circle/banner and a warning. The only issue with the FLARM output and EV4 seems to be a momentary loss of communication every 20 seconds or so.

Is it correct that FLARM protocol will output only a single bearingless target? Is it the closest one in terms of calculated horizontal distance or vertical distance?

b3nn0, 2020-05-12:
About protocol selection: Your workaround is correct and should work without issues.
Guess a switch would make sense. Pull request is welcome if someone wants to do it.

Yes, Flarm-NMEA can only output one target. IIRC, it's the closest target (horizontally), as long as it's closer than 15km away horizontally and within 2000ft vertically.

Krasnall, 2020-05-19:
> While waiting for b3nn0's reply, can I ask you which GDL90 Altitude switch setting you are using for Stratux Europe with EasyVFR? 

Answer from EasyVFR devs:
> It doesn't really matter, you can (should ;-))) always "calibrate" you device using menu->system, there is an option called "AMSL geoid correction".

VirusPilot, 2020-05-20:
> Answer from EasyVFR devs:
> 
> > It doesn't really matter, you can (should ;-))) always "calibrate" you device using menu->system, there is an option called "AMSL geoid correction".

Hm, I am afraid you may have to ask the question as follows: When processing the GDL90 data from any connected traffic device, is EasyVFR interpreting the "GDL90 Ownship Geometric Altitude" as Altitude Above Ellipsoid (HAE) as defined in the GDL90 spec or as Altitude Above Geoid(MSL)?

susanneeichel, 2022-07-09:
If it's only about bearingless targets, you can use globalSettings.EstimateBearinglessDist to disable sending them out via GDL90. Wouldn't that be enough to solve your usecase?

closed 2023-05-18 by b3nn0