stratux/stratux #44: second wifi usb dongle

Michal2184, 2020-05-12:
1. Stratux version: stratux-v1.6r1-eu015-27843b2e
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: anker 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Hi all

Iam not able to configure my usb wifi dongle.   Has been using stratux as pass thru internet dev and wifi usb connecting to my phone.  i was working fine on older versions but since direct wifi added it stopped. USB wifi D is beeing seen by stratux as iwconfig is showing wlan1 present. Wpa_supplicant.conf has been filled with correct details, routing tables has been set ,NAT also set ,but like shown below wlan1 is failing when ifup function is run.  
 
If i run "ifup" on wlan1 it gives: 
>sudo ifup wlan1 
>wpa_supplicant: /sbin/wpa_supplicant deamon failed to start
>run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
>ifup:failed to bring up wlan1
>

In  /etc/network/interface I did unchecked and edited as usual:
allow-hotplug wlan1
iface wlan1 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

Also it might be worth adding in future versions as i think a lot of people have tablet without internet ,but has a mobile that can run hotspot.  

thanks
Mike

b3nn0, 2020-05-12:
Two possible reasons.
1. The newer versions use wpa supplicant already to offer WiFi direct, if it is enabled.
This might conflict with your settings.
For your setup, you will need to use the hotspot mode (which I assume you already do).
2. Because of this, Stratux will also overwrite the /etc/wpa_supplicant/wpa_supplicant.conf file whenever you change the network settings in the web interface.
As a workaround, I recommend you to make a seperate file (e.g. /etc/wpa_supplicant/wpa_supplicant_client.conf), put your config in there and reference that from /etc/network/interfaces.

Or even easier, just configure the device directly in /etc/network/interfaces like this:
allow-hotplug wlan1
iface wlan1 inet dhcp
  pre-up sleep 5 # seems to be required in recent debian versions, don't know why
  wpa-ssid <your wifi name>
  wpa-psk <your wifi password>

Also keep in mind, that whenever you change the network settings in the web ui, the /etc/network/interfaces will be re-written by stratux and your changes are lost.
To keep your changes when changing stuff in the web ui, you might want to also place it in /etc/network/interfaces.template.

Michal2184, 2020-05-13:
Thanks a lot it worked!

 Indeed "pre-up sleep 5" was missing and Pi was refusing to let it run. So for anyone who would like to use it here is full guide how to do it: 

1. Edit /etc/network/interfaces  
Type:
> sudo pico /etc/network/interfaces

Inside file uncheck and add: 
>       allow-hotplug wlan1
>       iface wlan1 inet dhcp
>       pre-up sleep 5 
>       wpa-ssid  your_phone_hotspot_name
>       wpa-psk  your_wifi_password

2. Edit /etc/sysctl.conf   
>sudo pico /etc/sysctl.conf

Uncheck line:
>      net.ipv4.ip_forward=1

3.  Start NAT
>     sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"

4.  Set Routing over NAT by typing one by one:
>   sudo iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE
>   sudo iptables -A FORWARD -i wlan1 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
>   sudo iptables -A FORWARD -i wlan0 –o wlan1 -j ACCEPT

5.  Copy Routing into file 
>    sudo sh -c “iptables-save > /etc/iptables.ipv4.nat”

6.  Edit Stratux wifi setting in /usr/sbin/stratux-wifi.sh  so it can use copy of routing file that we created every time it restart. In file find lower down:
>sudo pico /usr/sbin/stratux-wifi.sh

Find these lines inside file and add one in this place :
>  sleep 2 
>  wLog "Restaring DHCP services"
>  /bin/systemctl restart isc-dhcp-server
>           iptables-restore < /etc/iptables.ipv4.nat         # ADD THIS LINE TO SET ROUTING
>}

7.  Reboot or shutdown device
> sudo shutdown now

Your Stratux from now will be connecting to your phone and share internet connection over all devices that you connect to it.  Also you don't need to connect your Stratux  to monitor or keyboard, I did it all via SSH client from my PC but you can use tablet as well. All you have to do is connect normally to stratux and in SSH connect to 192.168.10.1  with standard Pi login and password.  Wifi USB dogle cost me £4 from Amazon so great cheap upgrade to  Stratux! =

Many thanks for help!

closed 2020-05-13 by Michal2184

tmmtaha75, 2020-06-11:
Hello again: I don't know whether I should open a new issue or continue in this one. I followed your steps above to the letter. I even put screenshots of everything I typed in the terminal and the files I modified so you can see that everything in there matches the steps above. The dongle is connected to the internet and I can ping websites from the raspberry pi, but I can not get the internet to be forwarded to my iPad or my android phone when connected to the Stratux Wifi. The pics of the modified files and the steps in the terminal are in the following link:
https://drive.google.com/drive/folders/1-ZDHWY_5nnjk_lGwgUpfz8jxh7gDwn_U?usp=sharing

Any help would be greatly appreciated!