stratux/stratux #45: No SDR devices detected

russgreen, 2020-06-01:
1. Stratux version: 1.6r1-eu016
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [X] yes
      - [ ] no

    power source:

    usb cable: 

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Skydemon 

    EFB platform: (e.g., iOS 9.2) iOS 12.4.5

    EFB hardware: (e.g., iPad Mini 2) iPad Mini 2

4. Description of your issue:

After flashing to new version GPS is working but on the status page status is connected
[stratux.log](https://github.com/b3nn0/stratux/files/4710395/stratux.log)

Recent clients 0
SDR devices 0
Messages 
1090ES 0
OGN(disconnected) 0

b3nn0, 2020-06-01:
Sorry, can't reproduce the issue. Some notes:
- Make sure that the SDRs are connected _before_ booting the stratux
- After boot, you have to wait 1:30min until the SDRs are enabled

Can you enable debug logging in the settings and reboot, and post the log file here?
And, if you know how, maybe connect via ssh and post the output of the "dmesg" command.

russgreen, 2020-06-01:
SDR's are in OK first.

Log attached.

SDRs are the v2 stratux dongles.  It was a Pi 3B v1.2.  But just tried in a 3B+ and got the same result

[stratux.log](https://github.com/b3nn0/stratux/files/4711396/stratux.log)

b3nn0, 2020-06-01:
Strange.. there is no indication of your SDRs in the log file whatsoever.
For testing, I just re-imaged my SD card with this release and put it in my Pi 3B with Ublox GPS + 2 SDRs. Worked instantly without any issues.
Maybe re-image the SD card? Maybe something went wrong there.
Or do you have a spare SD card to test with (maybe it is broken)?
Do you know how to connect to your Stratux via SSH? If so, check "lsusb" if the SDRs are detected by the OS correcly. Output should have something like
"Bus 001 Device 006: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T"
for each dongle..

russgreen, 2020-06-01:
looks like my SDR's are the issue.  I put some nooelec ones in and they show up

b3nn0, 2020-06-01:
Strange. I'm using the Stratux v2 aswell. Yours worked with the previous release and now not any more? Can you post a photo of them?

russgreen, 2020-06-03:
It did work in the previous. But I flashed the card back to the previous version and now they're not working. Back to 1.6r1-eu016 with nooelecs and seem OK. Pic attached
[image]

b3nn0, 2020-06-03:
Very strange. These are indeed the exact same radios I'm using successfully on both versions..
Sorry, no idea what the issue  could be..

closed 2020-07-21 by b3nn0