stratux/stratux #46: Traffic no longer showing in Foreflight with 1.6r1-eu016 EXPERIMENTAL

tmmtaha75, 2020-06-02:
1. Stratux version: Stratux Europe Edition 1.6r1-eu016 EXPERIMENTAL
2. Stratux config: ADSB and UAT SDRs plus a SoftRF USB Dongle

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: External Battery

    usb cable: High load

3. EFB app and version: Foreflight 12.4, Skydemon iPad 3.13.6.238 (both latest versions)

    EFB platform: iOS 13.5

    EFB hardware: iPad Pro 10.5

4. Description of your issue: Stratux Europe Edition 1.6r1-eu016 EXPERIMENTAL does not show traffic at all in Foreflight, but works fine with Skydemon. The previous version of Stratux software works fine with both Foreflight and Skydemon. Skydemon shows ghost aircraft warnings with both codes for mode S and SoftRF dongle transmitter entered into the Stratux software (not specific to the latest version of Stratux software).

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-06-02:
For Foreflight: Nothing really changed in the way Stratux communicates with the EFB, but will check if it is reproducable.
wrt. ghost aircraft: That shouldn't really happen if properly configured.
Can you please
- Post a screenshot of the ghost aircraft in SD
- Post a Stratux log file after this occurs
- Post a screenshot of the settings in Stratux

tmmtaha75, 2020-06-04:
Thank you for your reply. All the files you requested can be downloaded at:

https://drive.google.com/drive/folders/1tGzrp6gn1Ug0RVwal9719RHBOeN3OxA7?usp=sharing

On a related note, I also reverted back to the previous version of Stratux "15," and traffic still cannot be seen with Foreflight. This makes me believe it may have something to do with the Foreflight recent app update. :(

b3nn0, 2020-06-04:
Okay now.. wrt. traffic reception: Just to make sure: You can only run one app connected via GDL90 at the same time. So when SD is using GDL90, ForeFlight can not do it at the same time on the same device (just because you posted a split-screen screenshot, it is expected to not work in this condition).
A workaround would be to change the Stratux IP to 192.168.1.1 and connect ForeFlight via GDL90 and SkyDemon via Flarm-NMEA.

Wrt. ghost aircraft:
What I see in your SD screenshot is the Mode-S bearingless targets . You can disable those in the settings. Can't see if that's from your aircraft - I assume there was a Mode-S only target near you at same altitude?
If it really is just you, please double check that the transponder hex in your settings is correct.
There is no complex logic going on for Mode-S exclusions - it's simply "same ID -> ignore".
There is no screenshot of SoftRF dongle ghosting. The Code for excluding this is quite a bit more complex, but since I'm also flying with this setup, it is also tested quite well. I can't see any exclusion-failures in your log.. are you sure the hex code is correct?

tmmtaha75, 2020-06-04:
wrt. traffic reception: Just to make sure: You can only run one app connected via GDL90 at the same time. So when SD is using GDL90, ForeFlight can not do it at the same time on the same device (just because you posted a split-screen screenshot, it is expected to not work in this condition).
I know that, that's why I'm using them in splitsceen now after Foreflight stopped showing traffic when it's running on its own (without Skydemon). Now that Foreflight is allowing splitscreen I'm using that as a workaround since most of my flying is IFR: Foreflight for navigation and Skydemon for traffic.

Wrt. ghost aircraft:
What I see in your SD screenshot is the Mode-S bearingless targets . You can disable those in the settings. Can't see if that's from your aircraft - I assume there was a Mode-S only target near you at same altitude?
The traffic you see is my mode S transponder code which stopped showing the ghost aircraft in Foreflight (when it was showing traffic) once entered in the exclusion list in Stratux settings. Skydemon shows the ghost mode S once in a while and not all the time. The Stratux traffic page confirms that was me, the only traffic being seen. 

If it really is just you, please double check that the transponder hex in your settings is correct.
There is no complex logic going on for Mode-S exclusions - it's simply "same ID -> ignore".
There is no screenshot of SoftRF dongle ghosting. The Code for excluding this is quite a bit more complex, but since I'm also flying with this setup, it is also tested quite well. I can't see any exclusion-failures in your log.. are you sure the hex code is correct?
Yes, the hex code is correct and worked before flawlessly. Like I said above, Skydemon only shows it sporadically.

And one last thing, I can't see OGN trasnmitters anymore. I used to see them when I am close enough. Added a pic in the Google Drive link above from Foreflight a couple of weeks ago showing OGN traffic in my area.

b3nn0, 2020-06-04:
Hmm, that's all very strange indeed.. I somehow can't believe that ForeFlight broke it - there are so many users using Foreflight with the original Stratux, which has the same code in this respect. If it was a general issue, there would be tons of people reporting it there as well.. However, I'm pretty much out of ideas in this regard..

The ghost issue is also very strange. I just tripple checked the code again for the Mode-S handling, and can't explain how it could fail - at least for Mode-S.
Again, for your Dongle, it is a different story - much more complex code for the ownship-detection when multiple codes are entered - but I have tested it quite extensively.
However, when Stratux does ownship detection and detects that one of the entered codes is received, but detected not to be really you, then it would log something like this:
`Skipping ownship %s because it's too far away (%fm, speed=%f, max=%f)`
or
`Skipping ownship %s because the altitude is off (%f ft)`
which I can not find in your log file..

What makes you think that you can't see OGN transmitters? Only OGNTP, or any of the 868Mhz protocols? Don't you receive your own dongle any more? [PAW is buggy in the beta and expected not to be received, but the rest should work fine]

tmmtaha75, 2020-06-04:
I'm not too much worried about the occasional ghost mode S in Skydemon as long as it's not showing all the time. 
As for the OGN issue, all aircraft at my home airport are equipped with OGN transmitters. I used to see them occasionally, and especially when I was in the traffic pattern (as in the pic in the Google Drive folder). Now, I see my own transmitter for a few seconds then I disappear (maybe because I entered my own SoftRF code in the exclusion list). I ordered another SoftRF dongle in case mine is now defective. Would you recommend the T-beam or any other transmitter/receiver to see OGN traffic? For me, the mode S warnings and receiving OGN are very important because of my job as a flight instructor in Hungary. The Stratux has helped me avoid a lot of sticky situations.

b3nn0, 2020-06-04:
No specific hardware recommendation from me. I have both, with SoftRF AND OGN Tracker software installed.
In the traffic list in the web interface, you should always see your dongle, that page does not consider the exclusion-IDs. The exclusions are only relevant for sending to the EFB.

You should still be able to see OGN trackers without issue. Note however that the range might be slightly reduced due to a new software stack for signal demodulation (but with the added benefit of drastically reduced delay) in eu016.

One more note: If you have a SoftRF/OGN Tracker connected to the Stratux, Stratux uses both, the traffic received by your SoftRF AND the traffic received by your 868Mhz SDR together. So if either one of them works, you should see OGN targets.

tmmtaha75, 2020-06-05:
I figured out the problem. For Foreflight, their recent update added a traffic option on the map that they advertised as a way to show internet FlightAware traffic in Foreflight. I had turned that off because that kind of traffic is useless for my purposes. It turned out that if you turn off that option, you won't see traffic in Foreflight with Stratux or any other connected device. I turned it back on and now traffic is visible.
I borrowed an OGN transmitter from a friend and placed it next to the Stratux to test the reception with different settings (my configuration is 1090 SDR, UAT SDR older version, SoftRF dongle, NO internal GPS and eu016 software):
(1) UAT on: In this case the SoftRF received the OGN signal, the UAT SDR was dedicated to UAT (no towers in Hungary but certainly useful abroad)
(2) UAT off: In this case I didn't quite know whether the SDR or the SoftRF was receiving the OGN signal, maybe both? But there was reception.
In both (1) and (2) I couldn't see my pwn SoftRF transmission, I only saw the OGN transmitter I borrowed. However, my SoftRF was clearly seen in my friend's Stratux (only 1090 SDR, UAT SDR, internal GPS and eu016 software) next to mine. The thing I can't explain is that sometimes I can see my own SoftRF in my Stratux but sometimes it won't show.
(3) When adding an internal GPS to the above configuration, the Stratux ignored the SoftRF GPS and used the internal one and the only way to activate OGN reception was to turn off UAT, otherwise OGN shows as "disconnected." But even with UAT turned off I couldn't see the OGN transmitter I borrowed!

Foreflight has a big flaw for European users: when connected to any ADSB device, Foreflight does not receive internet (assuming that US users can get weather and everything from their device which is not the case in Europe). With the latest Foreflight update, they finally enabled multi-tasking. This allows me to use Skydemon for traffic and Foreflight next to it for IFR flying. The big thing now would be the AHRS, as one would assume since Foreflight is not receiving traffic anymore because Skydemon is running next to it and getting the traffic that the AHRS won't work. Actually, the AHRS is being received by Foreflight because the iPad is connected to the Stratux Wifi and Foreflight is showing it as X-Plane input!

In conclusion, despite the ambiguity of not being able to see my own SoftRF transmitter in my own Stratux sometimes, the following would be the optimal configuration for IFR pilots in Europe:
Stratux without internal GPS but with SoftRF dongle (because you can transmit (be seen), 1090 SDR, UAT SDR and running in parallel to Skydemon which handles traffic and Foreflight getting internet via 4G at relatively low altitudes for weather radar and AHRS as X-Plane input, etc. Add an external ublox USB GPS and you can simply disconnect the SoftRF if you don't want to be seen and still utilize the AHRS and everything as above with the UAT deactivated to use that SDR for OGN reception.

tmmtaha75, 2020-06-05:
I want to confirm now that I couldn't see my own SoftRF in my Stratux due to the following:
When I changed the PPM from 100 to 1 and deactivated UAT, the internal SDR now was able to receive the SoftRF signal as it became dedicated to receive OGN. So something related to the PPM setting that was causing it not to be seen. So finally, with the PPM set to 1, I can see the SoftRF with UAT deactivated, and not see it when UAT is activated because that means the internal SDR isn't receiving OGN anymore and the SoftRF won't receive its own signal of course.

b3nn0, 2020-06-05:
Alright, so I suppose everything is working as expected, EXCEPT for the occasional short ownship-problem that you see your own transponder, correct?
Yes, the 868Mhz reception is fairly sensitive when it comes to PPM calibration and will only work if your PPM is correct to +-25.

tmmtaha75, 2020-06-05:
Alright, so I suppose everything is working as expected, EXCEPT for the occasional short ownship-problem that you see your own transponder, correct?
Correct!
I will keep you updated on my other tests. Thank you so much for your help and superb work!

closed 2020-06-05 by b3nn0