stratux/stratux #50: ICAO Hex not translated to show Reg for PaW traffic

wadoadi, 2020-06-06:
1. Stratux version:Stratux Europe Edition 1.6r1-eu013 with fix for issue #47 
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type: stratux internal

    AHRS
      - [x ] yes
      - [ ] no

    power source: Battey

    usb cable: Short 150mm 

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) IOS13  

    EFB hardware: (e.g., iPad Mini 2) iphone 11

4. Description of your issue:
In both Skydemon and on the stratux traffic page the the ICAO Hex address is shown but not decoded / looked up to display the Registration / tail number.

2020/06/06 21:43:05 GPS attitude: Average delta time is 0.10 s (9.8 Hz)
2020/06/06 21:43:05 New message: $GPRMC,204305.30,A,5244.45526,N,00033.00686,E,0.6,257.3,060620,,,A*55
2020/06/06 21:43:05 New message: $GPGGA,204305.30,5244.45526,N,00033.00686,E,1,5,1.00,6.4,M,45.8,M,,*6F
2020/06/06 21:43:05 New message: $GPGSA,A,3,,,,,,,,,,,,,1.0,1.0,1.0*33
2020/06/06 21:43:05 FLARM - ICAO target 405CA6 (405CA6) is 70.6 meters away at 277.9 degrees
2020/06/06 21:43:05 New message: $PFLAA,3,9,-69,1,1,405CA6!405CA6,0,0,0,0.0,8*59
2020/06/06 21:43:05 New message: $PFLAU,1,1,2,1,3,-82,2,1,70*71
2020/06/06 21:43:05 stats [started: 16 minutes ago]
2020/06/06 21:43:05  - Disk bytes used = 2.1 GB (29.1 %), Disk bytes free = 5.2 GB (70.9 %)
2020/06/06 21:43:05  - CPUTemp=49.39 [27.87 - 54.23] deg C, MemStats.Alloc=3.1 MB, MemStats.Sys=20 MB, totalNetworkMessagesSent=394
2020/06/06 21:43:05  - UAT/min 0/0 [maxSS=0.00%], ES/min 1/1, Total traffic targets tracked=1
2020/06/06 21:43:05  - Network data messages sent: 0 total, 0 nonqueueable.  Network data bytes sent: 0 total, 0 nonqueueable.
2020/06/06 21:43:05  - Last GPS fix: now, GPS solution type: 1 using 5 satellites (8/22 seen/tracked), NACp: 9, est accuracy 11.00 m
2020/06/06 21:43:05  - GPS vertical velocity: 0.46 ft/sec; GPS vertical accuracy: 22 m
2020/06/06 21:43:05  - Mode-S Distance factors (<5000, <10000, >10000): 2500.000000, 2800.000000, 3000.000000

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-06-06:
All 868Mhz traffic ids are resolved to tail number by checking the OGN device db. PAW targets that are in there are resolved. Is there a different database for PAW devices that is more complete?

wadoadi, 2020-06-07:
I don't know of a different source but it appears OGN are missing a lot of light aircraft. Hopefully must are flying with Mode S or ADSB out which broadcast the reg too.

b3nn0, 2020-06-07:
Don't know of any other database like this either. Resolving will work for aircraft that are in the DDB, or transmit their callsign via ADS-B.

closed 2020-06-07 by b3nn0