stratux/stratux #52: TTGO T-beam upgrade

Michal2184, 2020-06-07:
1. Stratux version:  stratux-v1.6r1-eu016
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Anker 10000

    usb cable: standard

3. EFB app and version: Skydeamon

    EFB platform: Android
    EFB hardware:  Huwaei M5

4. Description of your issue:

Hi
With new update I decided to buy TTGO T-beam board and BMP280 as would like to try it out. Don't think many people has done such an upgrade so guide is no were to be found. Can you shed some light how to get this working together? I presume t-beam need to be loaded with OGN tracker, than pairing it with BMP280? . Will it just work after plugin into stratux usb.....well i think it need a bit of tweaking?

Any sugesstions will be great!

b3nn0, 2020-06-07:
For a basic setup, you don't need to do anything special at all.
Pull out the GPS, plug in the T-Beam and you are good to go.
T-Beams often come preloaded with SoftRF - that will work fine in its default config. You can also switch to the ESP32 OGN Tracker software if you want.
BMP280 is only really needed if you don't have the AHRS module connected to your stratux already. Stratux will then, as a fallback, use the BMP280 connected to the T-Beam (with either software).

The only thing that _should_ be configured: enter the hex ID of your T-Beam in the ownship Mode-S exclusion list, because the SDRs will receive your T-Beam signal as well and show it as traffic if not excluded

Michal2184, 2020-06-07:
Thanks!  will get it tested this afternoon and come back with results.

Michal2184, 2020-06-07:
Just tested and no success yet. TTGO T-Beam board is newer edition with offset antenna to the right. Used correct data USB cable to connect with Pi board and TTGO has blue led and red led blinking so it has gps lock.  Stratux status page:

GPS hardware: FLARM (Not communicating)
GPS solution: Disconnected  

Board came with SoftRf already preinstalled but after few tries , I did flash it with current SoftRF and still no connection.  
i think iam missing something in stratux config...

b3nn0, 2020-06-07:
This indicates that Stratux has successfully detected that you connected a T-Beam. Good sign.
However, it doesn't seem to be able to receive data from it for some reason.
Have you tried
- rebooting the Stratux with the T-Beam plugged in at boot time
- Is the SoftRF set to output NMEA sentences (at least GNSS, potentially also Legacy)? It should be this way by default
- NMEA output via USB, no GDL90/dump1090 output?

If you know how, you could connect to the Stratux via SSH, run "stxstop" to stop the Stratux service, then cat /dev/flarm to check if NMEA is correctly arriving.

b3nn0, 2020-06-07:
You need to run "sudo -s" first to become root.
The output of cat /def/flarm should be lines starting with
$GNRMC, $GNGGA or similar. Not random gibberish. Potentially check the SoftRF web interface for the aforementioned configurations.

Michal2184, 2020-06-07:
worked with sudo -s ,but just one line with nonsense

root@raspberrypi:/home/pi# cat /dev/flarm
    @`            2     @  @

Also tried to log in directly onto t-beam and run app on tablet. It works and shows gps.

b3nn0, 2020-06-07:
Ok, that is not correct. Did you try the other steps?
- Reboot with T-Beam plugged in
- Verify settings in SoftRF web interface, as described above

Michal2184, 2020-06-07:
Ok just went on SoftRF web interface and in NMEA output you can choose

- off
- serial
-udp
-tcp
-bluetooth

GDL90: off 
Dump1090: off

So its missing USB therefore it dont work I presume

Michal2184, 2020-06-07:
[image]

b3nn0, 2020-06-07:
Ah, no. Serial is correct. Serial == USB in this case.
Did you already restart your Stratux? Again. **it has to be plugged in at boot time**.
I've just flashed SoftRF on my T-Beam and it works just fine in this case:
[image]

[image]

Michal2184, 2020-06-07:
SoftRF configurations are the same and it works so kind of puzzled now.
I tried restarting in various configurations. Stratux www status shows "FLARM" when t-beam connected ,and if i pull usb out it goes to "serial port not communicating". Again if i put it back in it goes to FLARM as on pictures below. 
[image]

After cable out: 
[image]

TTGO T-beam i have:
[image]

Do you use the same board?

b3nn0, 2020-06-08:
Yes, it is the same device (mine is rev 0.7, yours is rev 1.0, but that shouldn't make a difference).
No, I never put a battery in there. And no, you don't need any other hardware.
How does it work with eu018? In earlier versions you had to reboot Stratux for every disconnect/reconnect of the T-Beam. This should be more robust in eu018 (you should be able to re-plug/change hardware at any time and Stratux should pick it up after a few seconds).

Michal2184, 2020-06-08:
Updated to eu018.
I did try to reboot via stratux status "reboot option" while t-beam was connected and running, but no diffrence and still same FLARM (Not communicating) message.  There must be a reason why it work on 0.7 and not on 1.0 board.  
Ive noticed that cp210x driver on stratux is ver 1-1.3,  i did came across cp210x ver. 1-1.5

pi@raspberrypi:~ $ dmesg | grep ttyUSB
[    5.504806] usb 1-1.3: cp210x converter now attached to ttyUSB0

pi@raspberrypi:~ $ head -4 /dev/ttyUSB0

b3nn0, 2020-06-08:
I have the feeling that this is baud rate related..
SoftRF uses baud 38400 (which we also used to contact before eu016). For reasons, we changed that to 115200 lately - and it worked without issues here because of auto-baud. Maybe that doesn't work for you for some reason?
Can you, for testing, execute this:
stty -F /dev/flarm 38400 raw -echo
cat /dev/flarm

Michal2184, 2020-06-08:
tried and but still same problem.

pi@raspberrypi:~ $ stty -F /dev/flarm 38400 raw -echo
pi@raspberrypi:~ $ cat /dev/flarm

b3nn0, 2020-06-08:
Sorry, do all this as root please (sudo -s first).
Basically, do every command I tell you as root :)

Michal2184, 2020-06-08:
[image]

Does it not suppose to show numbers that make sense instead of what it shows?

b3nn0, 2020-06-08:
Yes.. when doing it right after pluggin in, it should look like this:
[image]

(it shows the T-Beams bootup sequence as well. Later, you will only see the $GP... lines).
There seems to be something strange going on with your T-Beam...

Michal2184, 2020-06-08:
[image]

I wought it worked. after i did run stxstart , checked status still not communicating, so i did reboot with t-beam plugged ,but sitll no joy.

b3nn0, 2020-06-08:
Would you mind joining the Stratux Slack group? I'll DM you there - makes it a bit simpler to communicate directly. You can invite yourself here: http://slack.stratux.me/

b3nn0, 2020-06-08:
Will be fixed in next version with proper baud auto detection.

closed 2020-06-08 by b3nn0