stratux/stratux #54: USB WiFi Dongle Problem

tmmtaha75, 2020-06-13:
1. Stratux version: Stratux Europe Edition 1.6r1-eu018
2. Stratux config: ADSB and UAT SDRs

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: internal

    AHRS
      - [x] yes
      - [ ] no

    power source: power source: External Battery

    usb cable: high power

3. EFB app and version: Foreflight 12.4, Skydemon iPad 3.13.6.238 (both latest versions)

    EFB platform: iOS 13.5.1

    EFB hardware: iPad Pro 10.5

4. Description of your issue:
Hello: I posted this on a closed issue but received no replies, so I deleted my previous post and opened a new one. I did the following steps (copied from the closed issue): 
Edit /etc/network/interfaces
Type:
sudo pico /etc/network/interfaces

Inside file uncheck and add:

  allow-hotplug wlan1
  iface wlan1 inet dhcp
  pre-up sleep 5 
  wpa-ssid  your_phone_hotspot_name
  wpa-psk  your_wifi_password
Edit /etc/sysctl.conf
sudo pico /etc/sysctl.conf

Uncheck line:

 net.ipv4.ip_forward=1
Start NAT
sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"
Set Routing over NAT by typing one by one:
sudo iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE
sudo iptables -A FORWARD -i wlan1 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan0 –o wlan1 -j ACCEPT

Copy Routing into file
sudo sh -c “iptables-save > /etc/iptables.ipv4.nat”

Edit Stratux wifi setting in /usr/sbin/stratux-wifi.sh so it can use copy of routing file that we created every time it restart. In file find lower down:
sudo pico /usr/sbin/stratux-wifi.sh

Find these lines inside file and add one in this place :

sleep 2
wLog "Restaring DHCP services"
/bin/systemctl restart isc-dhcp-server
iptables-restore < /etc/iptables.ipv4.nat # ADD THIS LINE TO SET ROUTING
}

Reboot or shutdown device
sudo shutdown now

I put screenshots of everything I typed in the terminal and the files I modified so you can see that everything in there matches the steps above. The dongle is connected to the internet and I can ping websites from the raspberry pi, but I can not get the internet to be forwarded to my iPad or my android phone when connected to the Stratux Wifi. The pics of the modified files and the steps in the terminal are in the following link:
https://drive.google.com/drive/folders/1-ZDHWY_5nnjk_lGwgUpfz8jxh7gDwn_U?usp=sharing

Any help would be greatly appreciated!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-06-18:
Not sure if it makes a difference, I have not tested it, but:
1: Usually, the lines belonging to a specific network interface in /etc/network/interfaces are indented, similar to how it is done for the wlan1 interface.
Please add spaces/tabs in front of the pre-up/wpa-* lines, but NOT in front of the allow-hotplug/iface lines. Again, not sure if it makes a difference, but that is how it is usually done.

2: You must have Smart-Wifi option _disabled_ for this to work. Note that when you change the setting, your /etc/network/interfaces file will be overwritten and you need to re-apply the changes.
[N.B.: the interfaces file is written from interfaces.template. If you want your changes to the file to survive wifi-settings changes on the web interface, you need to make the changes in there as well. However, that will then be reverted with the next Stratux update you do].

tmmtaha75, 2020-06-18:
1: Usually, the lines belonging to a specific network interface in /etc/network/interfaces are indented, ...
Done as per your recommendation (pic updated in the Google Drive link above for your reference) and still the internet isn't being forwarded.

2: You must have Smart-Wifi option disabled ...
It is disabled

EDIT: please note that these mods - while they work - are not supported nor endorsed by me. i.e. with these hints, you are on your own now:
I am now confident that this mod isn't working because even on the Reddit link that person posted it no one has confirmed it is. Would it be possible to direct me to a guide that is supported by you in order to have internet forwarding by a usb wifi dongle? This way I can at least try and ask questions in an area that has been verified by you and/or others.

b3nn0, 2020-06-18:
I can't, because that's not a setup I use or have tested. But I am certainly experienced with Linux network configurations and the described setup is reasonable and looks fine to me - but its easy to forget some detail when writing such a guide.
With some Linux knowledge, it should be easy to diagnose the problem (e.g. check if the IP-Tables file looks correct (cat  /etc/iptables.ipv4.nat), that the network interface config is fine (ifconfig), check the routing table looks ok (route -n), etc. But it's very hard to diagnose such problems without SSH access to the system.

b3nn0, 2020-06-18:
In fact, I think I know what the problem is. The Stratux DHCP Server doesn't provide the client with a DNS server.
Edit /etc/dhcp/dhcpd.conf and uncheck the `option domain-name-servers 4.2.2.2;` line, then run
`systemctl restart isc-dhcp-server`

tmmtaha75, 2020-06-18:
Thank you! It is working now and I'm writing this message using the Stratux hotspot!

closed 2020-06-18 by tmmtaha75