stratux/stratux #55: FLARM, OGN relative altitude

Koudy232, 2020-06-17:
4. Description of your issue:
Hello, I have been testing new release (eu18) and I am having two issues with OGN/FLARM reception.
1. Relative altitude is showed wrong on OGNs with baro sensor (OK-YUL 02 showing 425ft instead 800-900ft, the other one 24A8F0 is SoftRf  device with only GPS source and the altitude is correct 900ft)
2. HEX code filtering do not work if the device is in OGN database, if it no i the database filter is ok.

Please see atached pictures one is version eu13 and the second one eu18 with the some conditions.
[image]
[image]

b3nn0, 2020-06-17:
1) is already fixed in git and will work in the next version (will probably be released next weekend)
2) HEX filtering should work just as well, no matter if in OGN DDB or not. Please not that the targets on the traffic page of the web interface are NOT filtered. The filter only influences what is being sent to the EFB. The traffic page shows everything that is received. Note that eu013 wasn't able to receive all the protocols that eu018 can.

Koudy232, 2020-06-17:
Please see attached pictures. OK-YUL 02 (26A74C) is shown on both radar as well as SkyDemon and SoftRF (24A8F0) is not - both are filled as HEX code exception. In eu13 filter works for both devices.
[image]
[image]
[image]
[image]

b3nn0, 2020-06-17:
I have the feeling that this is related to the wrong altitude computation. Stratux EU has ownship detection and therefore notices when the target is "not really your own ship". One parameter that influences that is the altitude ("it is 500ft below, can't possibly be you"). Can you please post your log file? IIRC, Stratux will log when this happens, with information why it happens.

Koudy232, 2020-06-17:
Please find attached log file. Altitude difference could be the reason. Unfortunately all of my registered devices have baro sensor so the altitude difference is presented. 
['stratux.log'](https://github.com/b3nn0/stratux/files/4791739/stratux.log)

b3nn0, 2020-06-17:
Yes, that's the reason. See log file entries like this:
2020/06/17 10:57:01 Skipping ownship 26A74C because the altitude is off (756.169373 ft)

Therefore, this will also be resolved in the next version coming this weekend.

closed 2020-06-17 by b3nn0