stratux/stratux #56: OGN/Flarm altitude

Koudy232, 2020-06-21:
I have been testing new release (eu19) and still having issues with altitude reporting for devices with baro sensor. Now on ground altitude randomly changing from 0ft to 1500ft. Please see attached log file  (OGN with baro 26A74C). 
['stratux.log'](https://github.com/b3nn0/stratux/files/4809710/stratux.log)

b3nn0, 2020-06-21:
Hmm, just did some tests with a baro-equipped T-Beam, loaded with
- OGN Tracker firmware
- SoftRF Firmware set to OGNTP transmission
Both worked without issues.

Looking at your log file, it looks like 26A74C was indeed received at wildly varying baro altitudes:
"alt_std_m":362
to
"alt_std_m":146, 
within a second.

Can you please specify your exact software and hardware configuration? What Tracker device, what firmware, what equipment on the Stratux?

pjalocha, 2020-06-21:
I check the APRS data for thie OGN tracker as received by LKJA and indeed the pressure altitude reported is fluctuating widely. I would check the firmware of the OGN-Tracker, maybe it is a very old one or experimental or something else is wrong. if you need help to update the firmware ask me.
Thus this problem is more like the OGN tracker sends bad pressure altitude.

Koudy232, 2020-06-22:
Than you for the idea that problem could be OGN tracked device. It is STM32 OGN Tracker, but I am not sure about firmware version. I will do more testing tomorrow. Just to be sure till version eu13 Stratux works with only GPS altitude "alt_hae_m" is that correct? My Stratux configuration is "standard" - Raspberry Pi 3 Model B Board, V2 radios,  Stratux AHRS 2.0 - Sensors and Fan Controller.

b3nn0, 2020-06-22:
Yes, older versions used the GPS altitude and then tried to convert that to a barometric altitude by itself. This is suboptimal because the self-computed altitude could be off by 50-100ft. Therefore, a received baro altitude is prefered now - which should be better as long as your tracker actually sends it correctly.

pjalocha, 2020-06-22:
Let me know what is the hardware configuration of the OGN-Tracker: RFM69 or RFM95, BMP180, which GPS, I can prepare the firmware if needed.

Koudy232, 2020-06-23:
I have done some testing today and the problem is really OGN-Tracker. Tracker is one of the oldest version of DIY STM32 OGN Tracker with RFM69 module and **no baro sensor** so the problem is firmware. I realized that both tracking servers  the glidertracker and the live.glidernet.org presenting only GPS altitude so I haven't notice this issue until Stratux started using baro altitudes. Unfortunately most of the aircraft in our club and the surrounding clubs  as well are equipped with this OGN tracker version so it will take some time to replace them. Would you consider to implement the option to use only GPS altitude in new Stratux release. 100ft error is acceptable, but it is nice to have new protocols like PA and Fanet.

b3nn0, 2020-06-24:
I have just pushed a workaround for this. We now prefer the GNSS altitudes of received aircraft again and transform that to barometric internally. Only if your Stratux doesn't have a GPS fix, or no barometric altitude, the conversion can not be done and we fall back to the received baro alt (and if nothing is received, fall back to GNSS alt -> wrong altitude in this case).

Can you please  install this over the web interface and check if it works for you (unpack first of course)?
[update-stratux-v1.6r1-eu019-4ef62ebdb0.zip](https://github.com/b3nn0/stratux/files/4825010/update-stratux-v1.6r1-eu019-4ef62ebdb0.zip)

Koudy232, 2020-06-24:
New version has been installed and altitude presentation is the same - with mentioned OGN tracker on ground fluctuating from 0 to 1500ft. Please see attached log file.
['stratux.log'](https://github.com/b3nn0/stratux/files/4825583/stratux.log)

b3nn0, 2020-06-24:
Hmm, can you try this one and post log again if it doesn't work? Changed some details and improved logging.
[update-stratux-v1.6r1-eu019-b78c013c77.zip](https://github.com/b3nn0/stratux/files/4825724/update-stratux-v1.6r1-eu019-b78c013c77.zip)

Koudy232, 2020-06-24:
Now, it behaves the some way like SoftRF device with no baro sensor (24A8F0). The altitude is close to correct one (max error around 200ft with continuous changes but no fluctuation). Please see log file.
['stratux.log'](https://github.com/b3nn0/stratux/files/4826826/stratux.log)

b3nn0, 2020-06-24:
Guess that is to be expected when relying on GPS altitudes. Closing this.

closed 2020-06-24 by b3nn0