stratux/stratux #57: Pilotaware not filtered by ownship

wadoadi, 2020-06-21:
1. Stratux version:Stratux Europe Edition 1.6r1-eu018
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: Battery 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: the aircraft I fly has a built-in PaW stratux is set up with the ICAO Hex and Pilotaware is set with the same Hex however my own ship is shown as a red target.

is it possible for township filtering to ack upon Paw traffic too?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

wadoadi, 2020-06-21:
Let me try this again on my next flight, hopefully, Tuesday evening, as it looks like the ICAO did not save the change when I changed aircraft!

closed 2020-06-21 by wadoadi

b3nn0, 2020-06-21:
Just for information: PAW/OGN/... are handled exactly in the same way wrt. exclusions, so this shouldn't be an issue.