stratux/stratux #58: Connection Stratux to Foreflight as an X-Plane source

tmmtaha75, 2020-06-25:
1. Stratux version: 1.6r1-eu019
2. Stratux config: ADSB and UAT SDRs

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [x ] yes
      - [ ] no

    power source: External Battery

    usb cable: high power

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight 12.5

    EFB platform: (e.g., iOS 9.2) iOS 13.5.1

    EFB hardware: (e.g., iPad Mini 2) iPad Pro 10.5

4. Description of your issue:
Would it be possible to connect the Stratux to Foreflight as an X-Plane source knowing that Foreflight receives X-Plane data on UDP port 49002 or 49001? The following official Foreflight link shows that these are the ports used by Foreflight to receive X-Plane data:
https://foreflight.com/support/support-center/category/about-foreflight-mobile/204115525

Foreflight users have asked for years that Foreflight stops disconnecting the app from the device's cellular network when connected to an ADSB device. For European users, this is essential because we don't have UAT towers to receive weather data in flight but can pick up a cellular signal here and there to update that info. Foreflight when connected to X-Plane maintains cellular connectivity, which would solve all the current problems associated with connecting to a Stratux:
(1) We would get traffic and AHRS from the Stratux
(2) Everything else from the cellular network.

Any guidance would be appreciated and would take your amazing build to a whole new level with IFR pilots in Europe. Thanks in anticipation.
   
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-06-25:
And this is why I was so hesitant to include X-Plane 11 output in the first place. See the readme of this repository:

> X-Plane 11 compatible output for EFBs that support simulator input (**experimental, unsupported**. Might make it possible to connect Garmin Pilot). Based on original work by 0x74-0x62.

I do not want to support this feature. Use it if it works for you, if it doesn't feel free to provide code to make it work, because I will not.

Anyway, I will leave this open to see if anybody will step in.

One note though: The current implementation provides the data on UDP port 49002. It works fine with Garmin Pilot and SkyDemon (even though SD doesn't support simulated traffic, only ownship position).
So in theory, this should work without issues, except if Foreflight is doing something weird that I don't know of (and won't be able to find out because I don't even have it).

tmmtaha75, 2020-06-25:
Thanks for your reply and for leaving it open so others could pitch in with ideas/suggestions.

logix727, 2023-06-28:
it also does not work for me out of box on andorid garmin pilot latest it sees it and connects but no traffic /gps just ahrs

b3nn0, 2023-06-28:
I just made a commit bringing the X-Plane output a bit more in line with the real simulator. At least SkyDemon seems to like this a bit more (bug in SkyDemons implementation IMO). Maybe so does GP? I don't know, I don't have it. If you know how to compile yourself, feel free to try with the latest git version.

logix727, 2023-06-28:
ok compiled and gave it ago its the same issue.. No trafffic.... gps not available under connexant but do get green AHRS with simulator as source... no worries uh i know this is something u dont wanna debut imma try viruspilots build real quick to see any luck there

b3nn0, 2023-06-29:
Just did another update.. I'm fairly conviced that this should now run in Foreflight at least - I used their spec:
https://www.foreflight.com/support/network-gps/
not that this IS different from what X-Plane actually does, but I tested it in SkyDemon, Sky-Map and Enroute, and all seemed to work fine with that (Traffic, AHRS and GPS).

logix727, 2023-06-29:
Sweet will load and try on gp android should be same as ff

b3nn0, 2023-06-29:
Just found this, it says it works with GP. If you want, feel free to try to replicate this in xplane.go: https://github.com/astenlund/fs2ff/blob/master/DataSender.cs

It does look at least similar to the previous version, with the longer XATT message.

logix727, 2023-06-29:
That looks like it might make it work but then. Break the other ones he
fixed??  What to do from here is there a way to have both?

b3nn0, 2023-06-30:
- The other apps seem to be a lot more tolerant
- the other apps don't rely on this hack, they support GDL90 and/or NMEA

If you manage to find something that works with GP, I am happy to give it a try with the others. It might just work.

b3nn0, 2023-07-01:
Alright, so I borrowed another Android device and installed a GP trial version.
Observations:
- GPS from Stratux works fine in current master
- AHRS did not work, I fixed it and pushed.
- Traffic does not work.

So I installed X-Plane to see if it works with that, and no. Garmin Pilot just doesn't seem to support traffic from simulator at all. Not much to do there.

Would be happy if a foreflight user could do some testing with current master. If I don't hear anything, I will assume this is solved and close the issue.

closed 2023-07-08 by b3nn0