stratux/stratux #59: Frequent Stratux Disconnects after Updating from eu12 to eu19

michaelfuncke, 2020-06-30:
1. Stratux version: stratux-v1.6r1-eu019-52df50d6
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: Stratux GPYes 2.0 u-blox 8 GPS unit 

    AHRS
      - [x] yes
      - [ ] no

    power source: Ekrist Powerbank 26800mAh

    usb cable: Renkforce K-1470 Strom-Kabel Raspberry Pi 1,50m

3. EFB app and version: Foreflight v12.5 (56546)

    EFB platform: iOS 13.5.1

    EFB hardware: iPad Pro (12,9 inches, 3rd generation)

4. Description of your issue:

After updating from eu12 to eu19 Stratux "disconnects" to Foreflight at least several times per flight. WiFi connection is not lost. Rebooting the Stratux resolves the issue. Several minutes later (varying between 2 and 90mins) the disconnect happens again.
I have not changed any of the hardware and the issue never presented itself under eu12 or before that.
I have a feeling (which I cannot prove) that this happens again and again at the same rough locations (e.g. flying over the Ruhr valley in Germany).

[stratuxlog.txt](https://github.com/b3nn0/stratux/files/4852376/stratuxlog.txt)

b3nn0, 2020-06-30:
Sorry, can't really reproduce, but looking at your log file, it seems like the complete RasperryPi is rebooting (since the reboot is fairly quick, I assume you can't notice your WiFi disconnecting, as it immediately reconnects).
The only thing I can think of would be unstable power delivery (the eu019 version does use a bit more power to demodulate PAW/FANET). So it might be a weak power bank or USB cable. To debug this:

- Can you reproduce this at home on the ground (using the same power bank and cable)?
- Is the red LED on the RasPi continuously on? If not this indicates lack of power
- If you know how to connect via SSH, please do so after having it running for a while at home and run "dmesg" - any failures in power delivery should be logged there. Maybe even check the old log files (dmesg.0, dmesg.1.gz, dmesg.2.gz, ...) if you find power delivery issues in there.

michaelfuncke, 2020-07-03:
Thank you for your quick reply.
Apparently you are right. When I start up the Stratux the LED is solid red, then after a while it starts to flicker and after about 2 mins it turns off completely. Turns out it's my USB cable, which worked fine with the earlier not-so-powerhungry versions, but now doesn't anymore.
Thank you once again.

closed 2020-07-03 by michaelfuncke