stratux/stratux #62: AHRS V2.0 not calibrating

KallePG, 2020-07-08:
1. Stratux version: v1.6r1-eu019
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: External USB u-blox 8 GNSS

    AHRS
      - [x] yes
      - [ ] no

    power source: powerbank

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
This is a known issue discussed on reddit already. There is a problem with calibrating the new AHRS V2.0 sensors. The error message when trying to calibrate is:

>>>
Set AHRS Sensor Orientation: Failed!
There was an error: couldn’t read accelerometer: couldn’t determine biggest accel from 0.000 0.000 0.000
The orientation failed. Please try again.
<<<

There is an update for the cyoung-Stratux-version found at http://updates.stratux.me/builds. It will be rolled into the main image shortly. I would highly appreciate, if this update could be implemented into the European Version as well.

b3nn0, 2020-08-20:
Version EU020 contains the latest upstream code for the AHRS. I don't have the AHRS v2.0, so I can't test myself.
Can you update EU020 and re-test with that version=

KallePG, 2020-08-20:
Thanks for the update. I'll try to check it tonight and will let you know if it works.

 
 

Gesendet: Donnerstag, 20. August 2020 um 12:55 Uhr

wadoadi, 2020-08-20:
Where in Europe can you buy the AHRS v2.0?

----------------------------------------

KallePG, 2020-08-20:
As far as I know – nowhere. I purchased the AHRS in the U.S., during the short period of time, when they were available on amazon.

Unfortunately, there seems to be a compatibility issue with the published firmware. Or the hardware is faulty. It is a known issue, already being discussed on reddit. The drift is that high, that the AHRS is of no use during flight. But the barometer and the fan controller works. :)

KallePG, 2020-08-20:
AHRS V2.0 is working with the new firmware (EU020), like it was working in the updated cyoung-Version.

 

Unfortunately the AHRS V2.0 are either faulty or require a differently programmed firmware. That’s a known issue and probably the reason, why no AHRS V2.0 are for sale at present.

b3nn0, 2020-08-20:
Alright, I'll close this then for now until the situation becomes a bit clearer - apparently there are many DOAs with the the new AHRS module - we'll see how it turns out.

closed 2020-08-20 by b3nn0