stratux/stratux #63: Flarm Option always disables

morpheus2939, 2020-07-21:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ X] yes
      - [ ] no
      type:

    AHRS
      - [ X] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
ForeFlight

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
Hi I can not enable the Flarm option any more, as soon as I leave the page, it got disabled again.
The "raw" config file does not change as well.
Changing the 1090 and 978 Options to on or off works.

b3nn0, 2020-07-21:
This option has been renamed - the web interface you are using is outdated.
See the last sentence in the release notes of the more recent versions:
> Please make sure to clear your browser cache after installing.

Did you do this?

morpheus2939, 2020-07-21:
Thanks for the fast Reply!

 

 

And  sorry no I missed the brower cache clearing, as a programmer, I should rtfm better!

 

 

Now it works fine!

closed 2020-07-21 by b3nn0