stratux/stratux #69: Questions and Suggestions

c3o8k1g, 2020-09-27:
1. Stratux version: 1.6r1-eu021
2. Stratux config:
    SDR
      - [ ] single
      - [X] dual
    GPS
      - [X] yes
      - [ ] no
    AHRS
      - [X] yes
      - [ ] no
___________________________________________________________________________________________________________________________
**Questions and Suggestions:**
Not sure about the feasibility, but anyway, they would make it an almost perfect product, totally autonomous as well.
.

**Reception, Decoding and Displaying:**
- it would greatly help to have a diagram containing all the steps the information go through from reception to being displayed/relayed.
- apart from ADS-B, does it receive and decode pure FLARM, OGN or OGN-R?
- Bearingless targets - working criteria?
.

**FLARM/OGN data:**
- is it displayed together with ADS-B in Foreflight?
.

**Traffic advisories:**
   - Add support for a mini speaker and output of Radar page sound.
   - Add support for a [mini display](https://flarm.com/wp-content/uploads/2016/07/PowerFLARM-Portable-ADS-B.jpg)  /   [led indicator](https://flarm.com/wp-content/uploads/2017/08/flarm-igc-enl.jpg) much like a FLARM .
.

**Shutdown:**
   - more convenient shutdown command - linkable button to iOS home screen (like a bookmark that when pressed issues the shutdown command)?
   - even better, support for a push switch (not to cut power, but to issue the shutdown command)
.

**Explanation/Help menus**
- Improve explanation/help menus, especially regarding EU-specific functions and toggle buttons action.
- what are those sources marked with only a letter "e"?
.

**Ownship:** 
- drop down menu for multiple HEX, only one active at a time (fleet) - [same issue at cyoung/stratux](https://github.com/cyoung/stratux/issues/739#issuecomment-545686649), not relying on software to hide crucial information
.

**Update script**
- Does the update .sh file only affects the Stratux services or does it overwrite any changes to the OS? (root pwd, wifi config)
.

**Stat. page:**
It would be useful/interesting to have access to some graphs regarding the Rpi status, particularly regarding Temperature and fan operation.
Something along the line of either:
https://github.com/jprochazka/adsb-receiver/wiki/Performance-Graphs
https://github.com/XavierBerger/RPi-Monitor
.

**WiFi:**
- WPA2 instead of WPA (or is it because WPA2 requires more resources?)
.

**Internet connection - is the following correct?**
- Smart WiFi toggle - I could only find this reference and nothing more, does this apply to Stratux as well? 
"Stratus 3’s Smart WiFi feature, it’s telling the iPad that while WiFi is being used for short range data transfer, there is no internet connection. That allows the iPad to use its internal LTE radio (if available) while staying connected to Stratus."

- Multiple WiFi sticks - [WiFi client mode](https://github.com/b3nn0/stratux/wiki/WiFi-client-mode)
 . Raspberry Pi as passthrough for Internet data + inject Stratux data:
 . I am only worried about this:
 .  . - Application expecting Flarm-NMEA ... - what issue might it cause, say in Foreflight? Isn't stratux configured merge ADS-B and OGN/FLARM data in a single stream?
 .  . - Update may wipe everything?

b3nn0, 2020-09-27:
> it an almost perfect product

It is not a product, it is an open source project - you are as much in the position to make any of your recommended changes as I am. With the exception that you are the one who wants them, not me. So feel free to implement anything you want, make a pull request, write a wiki page, ...

> apart from ADS-B, does it receive and decode pure FLARM, OGN or OGN-R?

It receives any of the relevant protocols on the 868Mhz band, including everything that OGN receives + FANET + PilotAware. It receives directly. It doesn't use OGN-R.

> Bearingless targets - working criteria?

The target needs to be interrogated by a ground radar station from ATC. We are only listening to the responses passively.
If you are using the Flarm-NMEA protocol, it works OOTB. GDL90 doesn't support bearingless targets. An "emulation" toggle is in the settings which will place 8 aircraft in a circle around you to simulate the same.
Targets are of course only shown if they are estimated to be close enough and at similar altitude.

>  Is it displayed together with ADS-B in Foreflight?

All received traffic is forwarded to all connected devices on all protocols. Anything else wouldn't make much sense..

> Add support for a mini speaker and output of Radar page sound.

No interest from my side - do it yourself if you want it, I'm happy to include your pull request. Or pay someone to do the development (I'm available for contracting).

> Add support for a mini display / led indicator much like a FLARM .

You can combine Stratux with any traffic display that supports the respective protocols. You might want to take a look at the SoftRF SkyView for a cheap DIY solution. It works file with Stratux.

> more convenient shutdown command - linkable button to iOS home screen (like a bookmark that when pressed issues the shutdown command)?
> even better, support for a push switch (not to cut power, but to issue the shutdown command)

Do it yourself if you want it. But you can also just pull the power plug. I always do it and never had any issues.

> Improve explanation/help menus, especially regarding EU-specific functions and toggle buttons action.

Do it yourself if you want it.

> what are those sources marked with only a letter "e"?

This means they are none of ADSB, ADSR or TISB. Most likely Mode-S only. Depends also on what messages of a target are received yet and which not. But shouldn't really matter in practice.

> drop down menu for multiple HEX, only one active at a time (fleet) - same issue at cyoung/stratux, not relying on software to hide crucial information

Do it yourself if you want it. Stratux EU has already automatic ownship detection. You can enter multiple codes in the field (comma separated). Enter all aircraft you fly and Stratux will try to automatically detect which one you are actually flying (and still warn you for the others).

>  Does the update .sh file only affects the Stratux services or does it overwrite any changes to the OS? (root pwd, wifi config)

See this script to find out what files are included in the update.sh: https://github.com/b3nn0/stratux/blob/master/selfupdate/makeupdate.sh. Most of your system config should be safe, but not stuff that Stratux potentially needs updated from time to time (e.g. udev rules).

> It would be useful/interesting to have access to some graphs regarding the Rpi status, particularly regarding Temperature and fan operation.

Feel free to implement it then. Stratux displays temp on the main page, but now statistics are saved.

>  WPA2 instead of WPA (or is it because WPA2 requires more resources?)

It is actually WPA2. In fact, it is both - 1 and 2. The client device chooses (at least AFAIK).

> Smart WiFi toggle - I could only find this reference and nothing more, does this apply to Stratux as well?

Yes.

> Application expecting Flarm-NMEA

Stratux merges all traffic and makes "converts" it to GDL90, Flarm-NMEA and the X-Plane protocol. EFBs can choose whichever they want. Flarm-NMEA works in a way where the EFB actively makes a connection to 192.168.1.1:2000 and Flarm-NMEA is returned there.
However, if the Stratux is in client mode, it will probably not receive the IP 192.168.1.1, so your EFB can't connect in this mode. GDL90 connection can still be used, as it doesn't require an active connection from the EFB to the Stratux.

Update will usually not wipe anything.

If your questions are sufficiently answered, please close the issue.

b3nn0, 2020-09-27:
Oh, and by the way - even though I'm doing most of the development here, the intention of this fork was not that I develop Stratux-EU.
The intention was to manage a github repository where all the changes that people made for the European region could be collected, so that there is only one source, and not 10 different forks, each implementing one feature.
This is, why I only implement stuff that I want to have myself - and everyone else is expected to do so as well. If someone wants to make these changes available to a wider audience, he can make a pull request and I will distribute it in this version.

c3o8k1g, 2020-09-27:
I am sorry if my list was perceived like a request or a complaint. This was absolutely not my intention.

For the suggestions/feature portion, my idea was to put out the word to someone capable and interested in developing said functions.

As for the questions, I was simply asking for clarification, again to someone more knowledgeable.

I was not and am not expecting that you develop anything other than what interests you, that is what github is for, right?

Thanks.

Rubens-Baricello, 2020-09-27:
Hello Adrian,

 

thx for your work!

 

Thomas

wadoadi, 2020-09-27:
Has any one got Stratux working with SoftRF’s Skyview? If so what setting did you uses in both Stratux and Skyview?

 

Thanks

b3nn0, 2020-09-27:
> I am sorry if my list was perceived like a request or a complaint. This was absolutely not my plan. No intention to cause any malcontent.

No worries, I'm not mad or anything. There are just so many people with "suggestions" but no idea what the "modus operandi" of this project is. So sometimes my written words might seem a bit direct.

> Has any one got Stratux working with SoftRF’s Skyview? If so what setting did you uses in both Stratux and Skyview?

Yes, works without issues here. Configured SkyView as a wifi client with GDL90.
Also works via serial, but is a bit harder to set up there.

c3o8k1g, 2020-09-27:
> No worries, I'm not mad or anything. There are just so many people with "suggestions" but no idea what the "modus operandi" of this project is. So sometimes my written words might seem a bit direct.

No problem at all, I totally understand.
Would it be fine with you if I open an Issue where it is possible to share suggestions, so that anyone interested can find everything in a single place? 
Most of the times, something already exists but it is simply obscure until someone bring it up (like the SoftRF Skyview for me).

wadoadi, 2020-09-27:
Re: 

Has anyone got Stratux working with SoftRF’s Skyview? If so what setting did you uses in both Stratux and Skyview?

Yes, works without issues here. Configured SkyView as a wifi client with GDL90.
Also works via serial, but is a bit harder to set up there.

 

Fantastic as ever Adrian, I got my old Skyview out of the cupboard, never got it working with PaW or the main Stratux! tried the above and its working fine!

 

THANK YOU

closed 2020-09-27 by c3o8k1g