stratux/stratux #71: Shutdown link icon to home screen

c3o8k1g, 2020-09-29:
Hello @b3nn0,
after our conversation a couple of days ago, I started researching ways to shutdown the system safely.
I don't know how to send an improvement to the repository. In lieu, I created a [wiki here](https://github.com/cyoung/stratux/wiki/Shutdown-link-icon-to-home-screen).
Anyhow, here is the solution I found .
Although you might not be interested, others may be.

***
A more direct shutdown method, without the need to access the web interface.

Place the links you need in the home screen of your device (both Android and iOS). The links appear like any other app.

To add the links:
 - open the browsers menu 
 - select "Add to Home Screen"
 - assign an appropriate name

When you need to shutdown, simply press on the home screen button and the Stratux will shutdown.

**A blank page will open in a browser - this is normal.** After a few seconds, the device performs the required function.

Shutdown: http://192.168.10.1/shutdown

Reboot: http://192.168.10.1/reboot

The icons on the home screen act as push buttons triggering the above mentioned commands.

***
iOS only:

A more elegant way on iOS involves the creation of a custom profile with Apple Configurator loaded with the links and personalized icons.

The profile is then transferred to the device where the icons are then displayed.

[image]

[image]

Yellow = Reboot / Red = Shutdown

b3nn0, 2020-09-29:
Good solution.
The Wiki is just the right place to put information like that.
I just changed the project settings and I think you should now be able to edit Wiki posts here as well if you want to
( https://github.com/b3nn0/stratux/wiki ). If not, let me know.
As this doesn't really need any changes to the Stratux code, I think the issue tracker is the wrong location for information like this - or do you think differently/am I missing something?

c3o8k1g, 2020-09-29:
The wiki is the perfect place - [wiki created.](https://github.com/b3nn0/stratux/wiki/Shutdown-link-icon-to-home-screen)
Any chance to entirely delete this issue?
Thanks!

b3nn0, 2020-09-29:
Don't think so - closing is the right option. You might also want to move the "ideas" issue there, so others can modify/extend as needed, to create a general collection of ideas.

closed 2020-09-29 by b3nn0

c3o8k1g, 2020-09-29:
Done.