stratux/stratux #72: Stratux EU Diagram

c3o8k1g, 2020-09-29:
@b3nn0 

Please check the diagram for errors.

I created a wiki, if there is an error, let me know and I will rectify it.

Unfortunately, there is no other way to contact you directly.

Thanks.

[image]

b3nn0, 2020-09-30:
Would you mind joining the Stratux Slack? Would love to have a chat with you about it.
You can enter your E-Mail here: http://slack.stratux.me/
and you will receive an invite. I'll DM you when I see you join (if you use the same nick name)

wadoadi, 2020-09-30:
Nice diagram, I have the following observations, if I may!
Mode A  transmits an identifying code only.
mode C sends altitude
Mode S ICAO HEX, altitude, aircraft category,  capability, tail number and may be interrogated.
Modes S with ES enabled (ADS-B) adds position and may also add direction, speed... depending on the implementation.

In the diagram it suggest that mode S has position information which it doesn’t unless it is ES enabled I.e ADSB. Also it is possible to transpond mode A only. Where as I  believe Mode C is always A/C.

Thanks

Adrian

c3o8k1g, 2020-09-30:
Joined - same nickname.

closed 2020-09-30 by c3o8k1g