stratux/stratux #73: AHRS sensor GY-91 not working after version eu20

dns-007, 2020-10-05:
1. Stratux version: 1.6r1-eu20
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type: USB u-blox 7 GNSS receiver (NMEA-UBX protocol)

    AHRS
      - [ x] yes  - GY-91 
      - [ ] no

    power source: 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

After moving from version eu19 to eu20 the AHRS Sensor is no longer working. Also version eu21 has this issue for me. It is a DY-91 that I got in 2017 that always worked fine until version eu19. 
 stratux.log shows this:
2020/05/27 08:05:34  - stratux-wifi.sh - Running Stratux WiFI Script.
2020/05/27 08:05:34  - stratux-wifi.sh - Killing Hostapd services 
2020/05/27 08:05:35  - stratux-pre-start.sh - Running Stratux Updater Script.
2020/05/27 08:05:35  - stratux-pre-start.sh - Exited without updating anything...
2020/05/27 08:05:35  - stratux-wifi.sh - Stopping DHCP services 
2020/05/27 08:05:40  - stratux-wifi.sh - Restarting DHCP services
2020/05/27 08:24:41 Stratux v1.6r1-eu020 (8bd3bc503b6d8a997c05487cf081edb76ed9f9e8) starting.
2020/05/27 08:24:42 Entered uatReader() ...
2020/05/27 08:24:42 can't read settings /etc/stratux.conf: open /etc/stratux.conf: no such file or directory
2020/05/27 08:24:42 Developer mode set
2020/05/27 08:24:42 getDHCPLeases(): open /var/lib/dhcp/dhcpd.leases: no such file or directory
2020/10/05 09:46:17 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:18 GPS detected: u-blox NMEA position message seen.
2020/10/05 09:46:18 GPS attitude: Error calculating speed regression from UBX position messages
2020/10/05 09:46:18 GPS attitude: Error calculating speed regression from UBX position messages
2020/10/05 09:46:18 setting system time to: '20201005 08:46:18.100 UTC'
2020/10/05 09:46:18 Time set from GPS. Current time is 2020-10-05 09:46:18.105416979 +0100 BST m=+9.969315362
2020/10/05 09:46:18 setting system time to: '20201005 08:46:18.700 UTC'
2020/10/05 09:46:18 Time set from GPS. Current time is 2020-10-05 09:46:18.708399843 +0100 BST m=+10.328135778
2020/10/05 09:46:21 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:25 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:29 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:33 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:37 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:41 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:43 stats [started: 31 seconds ago]
2020/10/05 09:46:43  - Disk bytes used = 2.1 GB (6.6 %), Disk bytes free = 29 GB (93.4 %)
2020/10/05 09:46:43  - CPUTemp=36.48 [35.94 - 37.55] deg C, MemStats.Alloc=2.4 MB, MemStats.Sys=11 MB, totalNetworkMessagesSent=0
2020/10/05 09:46:43  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0, Total traffic targets tracked=0
2020/10/05 09:46:43  - Network data messages sent: 0 total, 0 nonqueueable.  Network data bytes sent: 0 total, 0 nonqueueable.
2020/10/05 09:46:43  - Last GPS fix: now, GPS solution type: 1 using 8 satellites (9/13 seen/tracked), NACp: 10, est accuracy 5.20 m
2020/10/05 09:46:43  - GPS vertical velocity: -0.27 ft/sec; GPS vertical accuracy: 10.4 m
2020/10/05 09:46:43  - Mode-S Distance factors (<5000, <10000, >10000): 2500.000000, 2800.000000, 3000.000000
2020/10/05 09:46:43 - Last IMU read: now, Last BMP read: now
2020/10/05 09:46:45 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:49 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:53 setting system time to: '20201005 08:46:53.200 UTC'
2020/10/05 09:46:53 Time set from GPS. Current time is 2020-10-05 09:46:53.20542126 +0100 BST m=+44.784327498
2020/10/05 09:46:53 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:46:55 setting system time to: '20201005 08:46:53.200 UTC'
2020/10/05 09:46:53 Time set from GPS. Current time is 2020-10-05 09:46:53.201105182 +0100 BST m=+47.348422655
2020/10/05 09:46:55 setting system time to: '20201005 08:46:53.700 UTC'
2020/10/05 09:46:53 Time set from GPS. Current time is 2020-10-05 09:46:53.702420026 +0100 BST m=+47.370708250
2020/10/05 09:46:53 setting system time to: '20201005 08:46:54.200 UTC'
2020/10/05 09:46:54 Time set from GPS. Current time is 2020-10-05 09:46:54.200962851 +0100 BST m=+47.389258431
2020/10/05 09:46:54 setting system time to: '20201005 08:46:54.700 UTC'
2020/10/05 09:46:54 Time set from GPS. Current time is 2020-10-05 09:46:54.700916833 +0100 BST m=+47.408478971
2020/10/05 09:46:54 setting system time to: '20201005 08:46:55.200 UTC'
2020/10/05 09:46:55 Time set from GPS. Current time is 2020-10-05 09:46:55.200996272 +0100 BST m=+47.427871249
2020/10/05 09:46:55 setting system time to: '20201005 08:46:55.700 UTC'
2020/10/05 09:46:55 Time set from GPS. Current time is 2020-10-05 09:46:55.700936875 +0100 BST m=+47.450168219
2020/10/05 09:46:56 setting system time to: '20201005 08:46:56.200 UTC'
2020/10/05 09:46:56 Time set from GPS. Current time is 2020-10-05 09:46:56.200906369 +0100 BST m=+47.768079392
2020/10/05 09:46:57 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:47:01 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:47:05 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:47:09 Could not identify MPU. v=c8, v2=73.
2020/10/05 09:47:13 stats [started: 1 minute ago]

If there is any help I can provide, let me know! Thank you!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-10-05:
This has most likely to do with the added support for the new Stratux AHRS v2, notably this commit:
https://github.com/cyoung/stratux/commit/7eeb32acf6fa31dc86ac94f95975e573f2619ae4
In previous versions, the I2C bus was just blindly used without confirming what it actually is.
In the new version Stratux tries to identify your AHRS Chip.

This works fine for the normal Stratux AHRS, but seems to fail for your specific GY-91. Stratux expects an I2C address of 0x71, but yours delivers 0x73.

Apparently, the I2C address of your GY-91 can be modified by pulling pins high/low, but I don't have such a device to test it.
In the meantime, we could add 0x73 as a detectable value and identify it as MPU as well.
- Are you familiar with SSH, light code editing and compiling stratux? That would make a test run simple.
- If not, are you familiar with connecting your Stratux to your home network for internet access and configuring a port forwarding in your router, so I can remotely connect to your Stratux? If so, contact me in the Stratux Slack please and we can discuss this.

dns-007, 2020-10-05:
Thanks for the quick response.
I am familiar with SSH, code editing and compiling. Actually, this is the first Flarm startux I started with in 2017 with the PepperJo code. 👍 
With the original Stratux AHRS sensor no longer available, we may have to look for a new AHRS chip as a replacement. If every DY-91 chip has a different I2C address, this is not user friendly. 
I do not have a problem to try to change the I2C adress of my chip, or buy a new one. 
Answer1: Barometric works, I get a pressure Alt
Answer3: I have this GY-91: https://www.amazon.de/gp/product/B017VT9TF4
How can we tell, if the GY-91 on the Internet is working with Stratux? What is supported?
Thank you!

b3nn0, 2020-10-05:
The GY-91 was reported to work fine, but I don't have one.
What did you do with the SDO/SAO pin?
"SDO/SAO - SPI Data output / I2C Slave Address configuration pin"
this might be what you have to play with (disconnect it if it is connected, pull it low/high for testing).
The GY-91 should have the correct address by default, but can be changed via this pin.

If this doesn't help, please try to try to modify sensors.go on your Stratux, line 158 reads:
```
} else if v2 == MPUREG_WHO_AM_I_VAL {
```
For testing, you might want to add an
``` || v2 == 0x73 ```
to that check and report back. But try the hardware solution first.
If neither works, try both at the same time (adding the check and play with the SDO/SAO pin)

dns-007, 2020-10-06:
Did some further testing. In the past the SDO/SAO pin is not connected. For testing I have now connected it it 5V pin. 
This is the i2cdetect with SDO/SAO unconnected:
root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --

This with SDO/SAO set high (5V). With this, both addresses are counted up and pressure an AHRS fails. 
root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77

Also, when set to high, I see this output:
2020/10/06 07:27:17 - Last IMU read: now, Last BMP read: now
2020/10/06 07:27:21 Error identifying IMU: remote I/O error
2020/10/06 07:27:25 Error identifying IMU: remote I/O error

So, some takeaways:
1. Why does the i2cdetect show 0x76, when the log says it is set to 0x73?
2. I would consider the PIN approach as failed. Agree?
3. I have ordered another GY-91 (https://www.amazon.de/gp/product/B07HMQZ7N9) and will test this one as soon as it arrives
4. Next I will try modifying the  sensors.go and I will register on Slack so we can chat there. 

Thank you!

b3nn0, 2020-10-06:
Looks like you are correct, yes. Very strange indeed.
Not connecting seems to be correct (as expected), since Stratux is expecting it at 0x68.

 Checking that specific register value seems to be a common approach to detect this chip, so it is a bit strange to me and I'm curious about the outcome of your investigation.

b3nn0, 2020-10-06:
The issue was caused by user not having an MPU9250, but MPU9255, which was then not detected any more.
Next version will also detect MPU9255 again. Closed.

closed 2020-10-06 by b3nn0