stratux/stratux #75: No Flarm with SoftRF dongle

myvilli, 2020-11-19:
1. Stratux version:Stratus Eu 1.6R 021
2. Stratux config:Sdr SoftRF GPS dongle

    SDR
      - [x ] single
      - [ ] dual

    GPS
      - [x ] yes
      - [ ] no
      type: SoftRF dongle

    AHRS
      - [ ] yes
      - [x ] no

    power source:external battery

    usb cable: high power

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
None: using Android tablet and browser
    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
I have updated the stratux EU to latest version (1.6R1 021), and the only option I have is the OGN (not connected) while I am using a SoftRF dongle as GPS and traffic receiver on USB port.
It was working flawlessly with previous version (018).

I am using a SoftRF dongle and baro sensor.
I have tried OGN protocol out on SoftRF, but it is not working either ( only GPS position seems to be taken in account through USB port)

Hope it helps.

Jeremy
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-11-20:
In the newer versions, the OGN(disconnected) only reflects the state of the SDR-OGN-Stack, not the SoftRF OGN reception.

SoftRF is not really handled like a traffic receiver in the code. It's basically handled like a GPS device, that happens to send us traffic messages from time to time that we incorporate.
In fact, these messages are used even if OGN is disabled in the settings.

Are you testing with actual traffic/a nearby sending device? Because that's the only way you can test this - the OGN(disconnected) is a bit misleading.

myvilli, 2020-11-20:
Thank you for your reply.

This is my setup, and how I tested it in previous versions:
(I was able to see the other device (Flarm) through my Soft RF tranceiver and GPS, and jetliners.)

My stratux was intercepting ADS-B messages via the SDR, and the SoftRF GPS wich include the LoRa modem was transceiving the Flarm messages In and Out to other Flarm devices around.
My stratux was able to detect both traffics : Jetliners around me, and the other "Flarm" device runing autonomously in my garden.

With the OGN "O" protocol, I cannot see the autonomous device.

I'll do more testing and report here, if there is anything new.

b3nn0, 2020-11-21:
> My stratux was intercepting ADS-B messages via the SDR, and the SoftRF GPS wich include the LoRa modem was transceiving the Flarm messages In and Out to other Flarm devices around.
My stratux was able to detect both traffics : Jetliners around me, and the other "Flarm" device runing autonomously in my garden.

This just still work, just like you describe. No matter if you enable or disable the OGN switch in the Stratux settings.
If you enable OGN in the settings, but have no SDR [not T-Beam, but real SDR] connected, it will show OGN (disconnected). But your Stratux will still receive the traffic provided from the SoftRF.

myvilli, 2020-11-26:
Hello,

After more testing, it just keeps working:
Stratux can sees Flarm through SoftRF, even without the OGN SDR. It was a mistake from my side.

This issue can be closed.

While OGN is activated, the Stratux can see also my own Flarm ID as an aircraft, it means SoftRF and OGN create ghosts.
I have tried to filter out my Flarm ID with the HEX Mode-S list, but it (obviously) does not work.

Is there any solutions to filter out some OGN traffics ?

BR

b3nn0, 2020-11-26:
> While OGN is activated, the Stratux can see also my own Flarm ID as an aircraft, it means SoftRF and OGN create ghosts.
> I have tried to filter out my Flarm ID with the HEX Mode-S list, but it (obviously) does not work.

This should work just fine.
Note however, that this filtering does not apply to the Stratux web interface. The web interface is more like a "raw data display" that contains everything and can't be filtered.
You should, however, not see yourself in an EFB (GDL90 or NMEA protocol) if you enter your SoftRF's HEX ID in the "Mode-S Exclustion list".

closed 2020-11-26 by b3nn0