stratux/stratux #76: Stratux as Aerodrome Traffic-Monitor

KBfly, 2020-12-03:
i think it would be a good idea if stratux monitored the aerodrome traffic like radar.
- on my latest stratux it's not possible to view the OSM Map (Radar). ->yes stratux ist online with internet. why? ...
- but is it possible to copy a offline map for example map of traffic ciruit etc. ... as overlay? -> this would be nice!
This would be a feature for a ground station yes, not in the air.

thank's for you great work! 

best regards an blue sky,
Karsten

b3nn0, 2020-12-03:
> on my latest stratux it's not possible to view the OSM Map (Radar). ->yes stratux ist online with internet. why? ...
The online maps don't depend on the Stratux having internet connection, but on the _client device_ having one. I.e. the device which you use to view the Stratux interface also needs to be able to access the rest of the internet (either by configuring the Stratux RasPi as a router or with a secondary network interface)

> but is it possible to copy a offline map for example map of traffic ciruit etc. ... as overlay? -> this would be nice!
This would be a feature for a ground station yes, not in the air.
Not possible right now in a simple way, since that would require quite a bit of work (understanding different GIS formats, georeferencing, reprojection of the raster image, ...).
For specific use cases, it is of course possible to modify the code manually, but solving that in a "generic" way is not an easy task.

b3nn0, 2020-12-21:
Closed due to no further response.

closed 2020-12-21 by b3nn0