stratux/stratux #77: Connecting u-blox GNSS module to ttyAMA0

rush0815, 2020-12-07:
1. Stratux version: 1.6r1-eu021
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: u-blox NEO-M9N   connected to ttyAMA0

    AHRS
      - [ ] yes
      - [x] no

    power source: 2A wall adapter

4. Description of your issue:
I am trying to bring up my NEO-M9N at the native ttyAMA0 port. Generally it is working, but is not recognized as a u-blox module and thus is not configurated like it should be. I mean thinks like update raten and all the other u-blox features that the "normal" GNSS modules don't have. 

What I have tried to do:
Since connecting the NEO via USB creates a symbolic link ttyACM0 -> ublox9, I tried to create this link manually while having connected the NEO to the ttyAMA0. That let stratux think a NEO-M9N module has been connected (see gps.go; line 218). The further reconfiguration seems to work online on the USB port, but not on the ttyAMA0.  I have compare the configurations from line 355-374 (file gps.go) to my settings in u-center. All seems to be identical. 

I ran a little test.
First, stopping stratix via stxstop. Then I connected my NEO-M9N via USB. The symbolic link to ublox9 has been created automatically. Then I started "cat /dev/ublox9" and got the following output:

$GBGSV,1,1,00,1*76

$GNGLL,,,,,,V,N*7A

$GNRMC,,V,,,,,,,,,,N,V*37

$GNVTG,,,,,,,,,N*2E

$GNGGA,,,,,,0,00,99.99,,,,,,*56

$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33

$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30

$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31

$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,4*36

While catting, I started stratux via "stxstart". The n9n has been detected and reconfigured immediately so that the cattting output changed to:

$PUBX,04,000033.00,101119,33.00,2079,18D,0,0.000,16*66
$PUBX,03,05,1,-,,,16,000,2,-,,,18,000,3,-,,,08,000,4,-,,,15,000,6,-,,,13,000*36
$PUBX,00,000033.20,0000.00000,N,00000.00000,E,0.000,NF,5303301,3750000,0.000,0.00,0.000,,99.99,99.99,99.99,0,0,0*28
$PUBX,00,000033.30,0000.00000,N,00000.00000,E,0.000,NF,5303301,3750000,0.000,0.00,0.000,,99.99,99.99,99.99,0,0,0*29
$PUBX,03,05,1,-,,,16,000,2,-,,,17,000,3,-,,,08,000,4,-,,,15,000,6,-,,,13,000*39
$PUBX,03,05,1,-,,,16,000,2,-,,,17,000,3,-,,,08,000,4,-,,,16,000,6,-,,,13,000*3A
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$PUBX,00,000033.50,0000.00000,N,00000.00000,E,0.000,NF,5303301,3750000,0.000,0.00,0.000,,99.99,99.99,99.99,0,0,0*2F
$PUBX,03,05,1,-,,,16,000,2,-,,,17,000,3,-,,,08,000,4,-,,,16,000,6,-,,,13,000*3A
$PUBX,03,04,1,-,,,15,000,2,-,,,17,000,3,-,,,08,000,6,-,,,13,000*16
$PUBX,03,04,1,-,,,15,000,2,-,,,17,000,3,-,,,08,000,6,-,,,13,000*16
$PUBX,03,04,1,-,,,15,000,2,-,,,17,000,3,-,,,08,000,6,-,,,13,000*16
$PUBX,03,04,1,-,,,15,000,2,-,,,16,000,3,-,,,08,000,6,-,,,13,000*17
$GNGGA,,,,,,0,00,99.99,,,,,,*56

I think the problem is, that when I fake a USB-connected module by manually creating the symlink "ublox9", stratux is reconfiguring the module and then waiting for messages that are only send via the USB output of the NEO and not the UART. 

How to deal with that issue ? 
One solution would be a simple textbox next to GSP on/off switch under Settings -> Hardware where the device can be configured. If I connect the NEO via USB, it would be "/dev/ublox9". By connecting via the native serial port I could change it to "/dev/ttymAMA0" and stratux would configure the module in that way that the correct message output would be pushed through the NEOs UART.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2020-12-08:
Please keep in mind that the PIs UART pins aren't known to be the most reliable things in the world, but what you can try is change the port configuration.

If you search in gps.go for UBX-CFG-MSG, you will find a nice table with configurations for each port.
And you will see that UART1 and USB are configured equally. I assume you are connecting to UART2 of the GPS. Feel free to add the respective 0x05, 0x01, etc. values there.

rush0815, 2020-12-08:
I am not sure if I really use UART2 becouse changing baudrate via u-center for UART1 results in unreadable stuff.
The integration manual

https://www.u-blox.com/sites/default/files/NEO-M9N_Integrationmanual_%28UBX-19014286%29.pdf

on page 23 says: "The  NEO-M9N  includes  two  UART  serial  ports.     UART1  can  be  used  as  a  host  interface  forconfiguration, monitoring and control."

You mentioned that I am maybe using UART2. I know that u-center has the possibility to configure this UART. But looking at the pinout of the module I can't find any second UART. No other u-block document seems to refer to any UART2. The only information that I could found is the the point that there is a second UART. Thats it! :-(

Do you have any informations how to connect to UART2? Or is the second UART connected internally to the USB so that UART2 = USB?

What do you mean exactly with "Please keep in mind that the PIs UART pins aren't known to be the most reliable things in the world..." Does the UART hang after a while or what happens ?

b3nn0, 2020-12-08:
Hmm, not sure what exactly is the state of M9N and UART. Might actually be the case that one of them is only used internally for USB. Not sure actually.

However, Stratux _does_ configure UART1 and USB identically, so _if_ you are using UART1, your symlink approach should work just fine.

You can actually confirm all that in U-Center:
Connect your UART Pins to the PI and monitor output. Start up U-Center and go to UBX->CFG->MSG. Select one of the options (e.g. PUBX0 or something) and check/uncheck the checkboxes while monitoring the output to see which output actually gives you what you want.

If it is UART1, everything should work already from what I can see. If it's UART2, you will have to make sure that UART2 is also configured like UART1 and USB.

"What do you mean exactly with "Please keep in mind that the PIs UART pins aren't known to be the most reliable things in the world..." Does the UART hang after a while or what happens ?"
Can't really tell you much more, this is only anecdotal knowledge. Heard of a few people that had problems with noise on the serial line, leading to lots of checksum errors, occasional disconnects. Could be that a ferrite around the cables would solve that, or that you don't have a problem at all. I really don't know and have not experimented with it myself. Feel free to try it, but maybe look out for problems related to that.

b3nn0, 2020-12-21:
Closed due to no further response.

closed 2020-12-21 by b3nn0