stratux/stratux #82: Add support for

Jixabon, 2020-12-17:
1. Stratux version: v1.6r1-eu021
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual
      978Mhz and 1090Mhz

    GPS
      - [x] yes
      - [ ] no
      type: Internal u-blox 8

    AHRS
      - [y ] yes
      - [ ] no
      AHRS 2.0

    power source: Anker 10,000mAh slim

    usb cable: Amazon Basics Short USB-C to Micro USB

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Foreflight v12.11

    EFB platform: (e.g., iOS 9.2) iOS 14.2

    EFB hardware: (e.g., iPad Mini 2) iPad mini 5 cellular

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Please add support for the new [UAT 978Mhz Low Power v3](https://www.amazon.com/Stratux-UATRadio-Low-Power-v3/dp/B07JNSHCLQ) that are now being sold with the Stratux kits. I didn't have an issue on the "US" version but it has a lack of development at the moment so I'm trying out this version in hopes that the handling of the new AHRS 2.0 chip is better. If it's not, at least more hope it will be better eventually due to more regular development.

All work is greatly appreciated!

b3nn0, 2020-12-17:
The AHRS implementation is the same as in upstream Stratux, so you will probably not see an improvement there.
The UAT Radio usage is also the same as in upstream and should therefore work just fine - i.e. it is supported already.
You just have to enable it in the settings (disable OGN, enable UAT).

closed 2020-12-17 by b3nn0

Jixabon, 2020-12-17:
So I did end up doing a road test to see how the EU does with the AHRS 2.0 chip and it is night and day compared to the US version. It is smooth and there is now lagginess past 20 mph. So there must be something different. I also noticed a much improved fix from the gps. Was only able to get below 12m on the US but got down to low single digits with EU. But not sure if that's due to the inactive UAT. There is only 1 reported SDR on the status page. When I physically remove the 1090 module it goes to 0. When I enable the UAT (OGN disabled) I noticed that the 1090 wasn't getting any messages which I was easily able to do before. So I disabled UAT again and boom 1090 was receiving messages. It seems as though UAT would have priority in this case and is trying to use the 1090 module to receive messages.

Jixabon, 2020-12-17:
There's also an issue filed in the base project about the v3 not functioning properly and that it is a known bug #791. That was opened in Sep 2019. Can't say that will be fixed anytime soon. Cyoung mentions a potential solution.

b3nn0, 2020-12-17:
wrt. GPS: Yes, we configure the GPS quite differently. Mainly to also make use of Galileo satellites.
wrt. AHRS: The AHRS code is really the same. But the GPS fix has quite some effect on AHRS quality, so maybe that's what helps here.
wrt. UAT: Oh wow, I think you are right. It worked fine in the past, but in a recent OS update, some udev handling seems to have changed. I had already updated the rules for GPS devices, but forgot it for UAT radio. Will be fixed in the next version (probably released around christmas). Probably nobody noticed because there are only very few UAT ground stations and aircraft in the EU (and most people here don't have a receiver either).

If you are familiar with the Linux shell, you can change your /etc/udev/rules.d/10-stratux.conf as shown in my commit and then reboot:
https://github.com/b3nn0/stratux/commit/142ee7a9c0cf44dd3e50fce066cb3e77b4e2012b#diff-4f3ea1f2edebc1bd276d4bdfefeb18068eccdaac22b70ea8406751a48c25acc8

If not, you will have to wait for the next version.

Jixabon, 2020-12-17:
BEAUTIFUL! Thank you so much! I'm not sure why the AHRS 2.0 performance is so much different but your work on here has actually made this project usable for me! I'm now reporting 2 SDR with UAT enabled and receiving 1090 messages. Will have to wait to test the UAT in the air. The issue I mentioned before may still apply. GPS is still a lot better than on the US version. Doesn't seem like the UAT being "on" is causing any more interference.