stratux/stratux #92: Unable to configure OGN T-Beam through NMEA in minicom

Shredder, 2021-01-24:
1. Stratux version: latest stable
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: TTGO T-Beam 0.7

    AHRS
      - [X] yes
      - [ ] no

    power source: Anker power bank

    usb cable: yes

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I have flashed my TTGO T-Beam version 0.7 with the OGN Tracker firmware which comes with this Stratux release on my Raspberry Pi. I have tested it in flight, and OGN shows my signal, so that works.

However, I want to fine-tune the OGN configuration beyond what the Stratux GUI offers, through NMEA commands. Here is my issue:

When I connect it to my Windows 10 computer vis USB (I have the requisite drivers installed), I can connect to it in minicom (under WSL) or Putty and see the stream of NMEA messages - all looks good. However, in both applications the device ignores control commands (Ctrl-C, Ctrl-X, Ctrl-V) so that I am unable to configure the OGN firmware through the serial NMEA console. I have tried long-pressing or rapidly entering the control commands, with and without flow control, but there was never any response by the device. It seems to me that it simply ignores any input.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-01-25:
Yes, this is expected. Because we configure the OGN Tracker to "UBX Pass-Through mode", it will disable Ctrl+* commands, since they can interfere with it.
Configuration with $POGNS NMEA should still work - Stratux uses that as well to configure the tracker.
You can, of course, also compile the firmware yourself with features and configuration you want.

closed 2021-01-25 by b3nn0

Shredder, 2021-01-27:
> Yes, this is expected. Because we configure the OGN Tracker to "UBX Pass-Through mode", it will disable Ctrl+* commands, since they can interfere with it.
> Configuration with $POGNS NMEA should still [...]

I have managed to compile `esp32-ogn-tracker` from source using this repository's `ogn/create_stratux_fw.sh` script. (I'm using this script to stay as close as possible to your pre-compiled firmware, which works for me.)

Could you please suggest which options in `config.h` to modify to make the device (a standard T-Beam version 0.7) responsive to Ctrl commands? Is it `WITH_GPS_UBX_PASS`?

b3nn0, 2021-01-28:
WITH_GPS_UBX_PASS is the one. But if you disable that, then Stratux will not be able to configure the GPS any more, which has some drawbacks. I would not recommend it - why do you need that anyway? You can do all $POGNS configuration, just the Ctrl+... commands don't work.
You could also just enable WIFI/Web Interface and configure that way.