stratux/stratux #93: Using both external GPS and TTGO T-Beam simultaneously

Shredder, 2021-01-24:
1. Stratux version: latest stable release
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: TTGO T-Beam and u-Blox 7 based

    AHRS
      - [X] yes
      - [ ] no

    power source: Anker power bank

    usb cable: yes

3. EFB app and version: (e.g., WingX Pro7 8.6.2) SkyDemon latest stable version

    EFB platform: (e.g., iOS 9.2) Android

    EFB hardware: (e.g., iPad Mini 2) Amazon Kindle

4. Description of your issue:

When I simultaneously connect both my external GPS mouse (u-Blox 7 based) and my TTGO T-Beam, Stratux seems to ignore my T-Beam. Got instance, no device is created for it in /dev and the web UI does not show configuration options for the T-Beam.

As I suspect that my) the GPS mouse is better than the GPS of my T-Beam, it would be nice if Stratux would be able to utilize both at the same time.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-01-25:
Due to the way GPS handling is implemented in Stratux, multi-GPS-handling is not a simple task and requires quite a bit of work - I have no interest to do that myself. If someone wants to do it and PR, I'm open to it.
However, normal USB GPSes are usually preferred over the T-Beam. So you can certainly have both plugged in. But you will - as you described - lose the configurability.

If you don't have to reconfigure frequently, you could just configure the Tracker once and then let it sit there for transmission. For reception, you still have the SDRs, so the T-Beam<->Stratux connection is mostly optional.

JNSwanson, 2021-06-23:
I have a T-BEAM with the later GPS module, and I found the reception not too good.  The backup battery has only 1mAh capacity and takes ages to charge, so it mostly has to do a cold start.  I replaced the GPS antenna with this one https://www.ebay.co.uk/itm/174725227219 whic works WAY better.  It now works indoors and has 20+ satellites in the solution.
Jeremy.

rvt, 2022-08-12:
We are currently working on having multi-GPS possibilities.
This shoudl allow to connect the t-beam and a GPS pug connected on stratux

[image]

jbrikmann, 2025-05-18:
Are you still working on this enhancement? I would be interested in it (both in helping in development and using it later)...

rvt, 2025-05-18:
Hey @jbrikmann ,

No I stopped developing with stratux and started to work on a other project with similar capabilities