stratux/stratux #94: Issue with LilyGO T-Motion device

MisterT2, 2021-01-26:
1. Stratux version: 1.6r1-eu022
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: VK-162 G-Mouse

    AHRS
      - [ ] yes
      - [X] no

    power source: external

    usb cable:

3. 

4. Description of your issue:
I'm trying to flash my S76G LilyGO T-Motion device, as recommanded in the Wiki.
Unfortunately, whatever method I'm using (from Linux or directly from my Stratux), I always get the same issue:
```
esptool.py v2.7-dev
Serial port /dev/ttyACM0
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

I may be wrong, but I guess this is linked to the fact the firmware fetched is for ESP32, while my S76G is using a STM32.
Correct ?

In that case, which firmware should I use (the SoftRF firmware in place does not seem to be properly recognized by Stratux-EU)

Thanks !

b3nn0, 2021-01-26:
OGN Tracker does not support the T-Motion. It is meant for the T-Beam.
You can use SoftRF just fine and it should work. Make sure to enable NMEA output and _not_ dump1090 or GDL90 output in SoftRF.

closed 2021-01-26 by b3nn0

MisterT2, 2021-01-26:
OK. Will buy T-Beam then... But I will try what you propose also :-)
Thanks !