stratux/stratux #95: Feature request:  option to use adsbexchange.com as data source

rush0815, 2021-01-30:
Hi there...

 Do know adsbexchange.com? It is „ the world’s largest co-op of ADS-B/Mode S/MLAT feeders, and the world’s largest public source of unfiltered flight data. “

I spend a lot of time on this site watching VFR  traffic in my regio. Their MLAT  is really awesome! For VFR pilots it is more important to know the other VFR traffic in similar altitudes than something else 10000 ft higher.

According to this I thought it would be nice to have their data as source and not the received ADSB stuff.
The advantages would be:
- no need for any SDR connected to the Pi. Thus less thermal problems since the SDRs are getting very hot
- longer battery lifetime because no SDRs have to be connected
- Much better accuracy for airplanes with Mode S bearingless transponders
- less hardware effort -> smaller device; Maybe a Pi zero would to the job quite well?

The one disadvantage is that you need to be connected to the GSM network and thus have a limited altitude for flying.

Of course the API is not for free, but it costs ony a few USD per month. The pricing can be viewed here:

https://rapidapi.com/adsbx/api/adsbexchange-com1/pricing

Some more links with info:

https://www.adsbexchange.com/data/#

https://www.adsbexchange.com/version-2-api-wip/

By the way: I am not involved in any way into adsbexchange.com. I am just fascinated that their service is that precise and unfiltered and love to use their website.

So, what do you think about this idea?

b3nn0, 2021-01-30:
Sorry to put it that directly, but I think there is no use to this whatsoever.
The quality of such traffic information is discussable due to connection delay and drops, etc. but even if we assume that it works,
there is no reason whatsoever to have Stratux do that.
You usually already have a Phone or Tablet up there (where your EFB is running), which has a GSM modem AND a Sim card.
You could just run a piece of software on there that provides a GDL90 stream to an EFB app. You don't need the Stratux for that at all. In fact, a Stratux would just make that more cumbersome since you need additional hardware, a power source, a second sim card, ... Or am I missing something?

rush0815, 2021-01-30:
Hm... yes you are rught. I did not think so simply :-D

Would it be possible that you send me an dump of the GL90 stream that stratux send to the EFB?

b3nn0, 2021-01-30:
Don't need me to do that.
You can just connect to the Stratux wifi and connect to it on UDP port 4000 and save the output somewhere (e.g. with tcpdump).
The specification is also publicly available: https://www.faa.gov/nextgen/programs/adsb/archival/media/gdl90_public_icd_reva.pdf

closed 2021-01-30 by b3nn0