stratux/stratux #96: WiFi Mode. Wifi-Direct, connection via WPA not working on iPad

rush0815, 2021-02-02:
1. Stratux version: v1.6r1-eu021
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ x] yes
      - [ ] no
      type: Neo-M9N

    AHRS
      - [ x] yes
      - [ ] no

    power source: 2A

    usb cable:

3.    EFB hardware: iPad Air 2020

4. Description of your issue:
 Current WiFi Settings:

WiFi Mode:    WiFi-Direct
WiFi SSID:   stratux
WiFi Passphrase :  xxx
WiFi-Direct Pin: 0000
Smart WiFi:   switch off
Stratux IP Address:  192.168.10.1

Tried to configure WiFi-Direct to have Internet connectivity on my EFB. 

As you said in a post on ulforum.de (https://www.ulforum.de/ultraleicht/forum/2_technik-und-flugzeuge/8050_stratux-europe-edition/seite-69.html), configuring stratux to WiFi-Direct will give me two different ways to establish a connection to stratux. The first one is via WiFi-Direct on Android and the second is the common WLAN connection via WPA. 
Both work on my Android phone, but no one of them works on my iPad. Apple does not support WiFi-Direct like Android, right ?

Trying to connect via WPA results always in a "wrong password" message. But the Passphrase is correct since I use it on the Android and it works.  Did I miss something?

If I understand it correctly: There is no way to establish a internet connection while the WLAN is used by my iPad with the EFB?

b3nn0, 2021-02-03:
Don't have an IOS Device to test, but this is what I know/heard:
- IOS does support WiFi Direct, but requires an 8-digit pin. A 4 digit one will not work
- The Passphrase to connect to the stratux-DIRECT network is the same as the WiFi Direct pin, but again, it needs to be 8 digits to work, since a WPA2 passphrase can't be only 4 digits
- To get internet connection while connected to the Stratux on iOS, you don't need to use Wifi direct at all. You can just enable the "Smart WiFi" option and use a normal AP. The Wifi-direct feature is mainly a workaround to get something similar on Android.

Does this help?

rush0815, 2021-02-03:
I changed both the Passphrase and the Direct Pin to "00000000" but connecting to DIRECT-stratux still results in a "wrong password" message.

b3nn0, 2021-02-04:
Then there is apparently something strange with iOS - I can connect to that Wifi with any device I have. No idea, sorry.
But the other option - normal hotspot and Wifi Smart enabled should work just fine.

Schildaero, 2021-03-19:
> Then there is apparently something strange with iOS
I can confirm that, but I cannot describe what exactly. I have here for testing besides an Android an iPad mini2 with 12.5.1 and an iPad mini4 with 13.4.1.
Couple of months ago I also had a really hard time to get these iOS connected. As I remember I was not able to connect to these with the direct-wifi at all. Nowadays (I just checked it) I don't have any issues to connect both iOS to the direct.wifi now. All is good. So maybe there was an iOS update, that fixed something (maybe there was an update of stratux/raspbian?!?) or maybe my devices just gut used to it.

Only tip I can give you right now is to consider install updates, if there are any available (and on iOS before doing so, maybe check if your liked apps will still work afterwards, as I heard there might be some suprises). Also start with trying a connection with the stratux in AP mode instead of Direct. This should work. Maybe try something different than just 0's. Maybe that's somethin iOS is not happy about.

From my understanding, on iOS the direct-wifi protocol is used for their AirDrop stuff and not the way Android does (allowing two simultaneous wifi connections.

I also had trouble connecting iOS to the softrf PrimeMarkII, as the iOS was unable to retrieve an IP Adress, whereas with Android it was never an issue.

Hope there is something that helps you.

@b3nn0 :

> Don't have an IOS Device to test
If there is something you need to be tested on iOS in future, please feel free to contact me.

b3nn0, 2021-03-20:
FYI, the next version will have a few minor changes to this.
1. the classic DIRECT-... AP Wifi will now be forced to 2.4G intead of 5G Wifi on the PI4
2. The Wifi Direct one will now advertise as "Network Infrastructure/AP" device, instead of a "Computer/PC" device.

I doubt it will make a difference, but who knows..

chmorgan, 2025-02-07:
Closing due to inactivity. Please reopen if this issue persists.

closed 2025-02-07 by chmorgan