stratux/stratux #101: No Pressure Altitude with BME 280

Quax-der-Bruchpilot, 2021-02-06:
1. Stratux version: VK-162 G-Mouse
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: VK-162 G-Mouse

    AHRS
      - [ ] yes
      - [X] no

    power source: Outlet in living room

4. Description of your issue: I have attached a BME 280 (no AHRS, just the BME) but I don't get a pressure reading. The ALT tab remains red. What I did: Checked the I2C Adress of the Sensor: 0x76. Checked the sensor itself on a Node MCU with an OLED screen and it seems to work. I double and triple checked the wiring on the Raspi 3B. I verified that in the settings baro sensor is enabled. Any suggestions?

2021/02/06 18:17:52 Logfile truncated
2021/02/06 18:17:53 setting system time from 20210206 18:17:53.359 to: '20210206 18:17:53.100 UTC'
2021/02/06 18:17:53 Time set from GPS. Current time is 2021-02-06 18:17:53.103089166 +0000 GMT m=+4.350636925
2021/02/06 18:17:53 GPS detected: u-blox NMEA position message seen.
2021/02/06 18:17:53 setting system time from 20210206 18:17:53.352 to: '20210206 18:17:53.600 UTC'
2021/02/06 18:17:53 Time set from GPS. Current time is 2021-02-06 18:17:53.600917656 +0000 GMT m=+4.610153487
2021/02/06 18:17:55 handleRestartRequest called
2021/02/06 18:17:55 signal caught: terminated - shutting down.
2021/02/06 18:17:55 Entered UAT shutdown() ...
2021/02/06 18:17:55 UAT shutdown(): calling u.wg.Wait() ...
2021/02/06 18:17:56 UAT read(): shutdown msg received...
2021/02/06 18:17:56 UAT shutdown(): u.wg.Wait() returned...
2021/02/06 18:17:56 UAT shutdown(): closing device ...
2021/02/06 18:17:56 UAT shutdown() complete ...
2021/02/06 18:17:56 Entered ES shutdown() ...
2021/02/06 18:17:56 ES shutdown(): calling e.wg.Wait() ...
2021/02/06 18:17:56 ES read(): shutdown msg received, calling cmd.Process.Kill() ...
2021/02/06 18:17:56 kill successful...
2021/02/06 18:17:56 ES shutdown() complete ...
2021/02/06 18:17:57 datalog.go: Starting data log shutdown
2021/02/06 18:17:57 datalog.go: Waiting for shutdown signal from dataLog()
2021/02/06 18:17:57 datalog.go: dataLogWriter() received shutdown message with rowsQueuedForWrite = 142
2021/02/06 18:17:57 datalog.go: dataLog() received shutdown message
2021/02/06 18:17:57 datalog.go: dataLog() has closed DB in /var/log/stratux.sqlite
2021/02/06 18:17:57 datalog.go: Data log shutdown successful.
2021/02/06 18:02:57  - stratux-pre-start.sh - Running Stratux Updater Script.
2021/02/06 18:02:57  - stratux-pre-start.sh - Exited without updating anything...
2021/02/06 18:17:57 Stratux v1.6r1-eu022 (16b446a6e856bc430b76b514061ec443326f3e55) starting.
2021/02/06 18:17:57 Entered uatReader() ...
2021/02/06 18:17:57 read in settings.
2021/02/06 18:17:57 Developer mode set
2021/02/06 18:17:57 client connected: 192.168.1.10:4000 ().
2021/02/06 18:17:57 client connected: 192.168.1.10:2000 ().
2021/02/06 18:17:57 client connected: 192.168.1.10:49002 ().
2021/02/06 18:17:57 datalog.go: Watchdog wants to START logging.
2021/02/06 18:17:57 client connected: 192.168.1.11:4000 ().
2021/02/06 18:17:57 client connected: 192.168.1.11:2000 ().
2021/02/06 18:17:57 client connected: 192.168.1.11:49002 ().
2021/02/06 18:17:57 datalog.go: dataLog() started
2021/02/06 18:17:58 ===== ES Device Serial: stx:1090:0 PPM 0 =====
2021/02/06 18:17:58 Entered ES read() ...
2021/02/06 18:17:58 Executed /usr/bin/dump1090 successfully...
2021/02/06 18:17:58 ===== UAT Device Name  : Generic RTL2832U OEM =====
2021/02/06 18:17:58 ===== UAT Device Serial: stx:978:0=====
Found Rafael Micro R820T tuner
2021/02/06 18:17:58 	GetTunerType: RTLSDR_TUNER_R820T
2021/02/06 18:17:58 	SetTunerGainMode Successful
2021/02/06 18:17:58 	SetTunerGain Successful
2021/02/06 18:17:58 	GetTunerGain: 480
Exact sample rate is: 2083334.141630 Hz
[R82XX] PLL not locked!
2021/02/06 18:17:59 	SetSampleRate - rate: 2083334
2021/02/06 18:17:59 	GetSampleRate: 2083334
2021/02/06 18:17:59 	GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2021/02/06 18:17:59 	SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2021/02/06 18:17:59 	SetCenterFreq 978MHz Successful
2021/02/06 18:17:59 	GetCenterFreq: 978000000
2021/02/06 18:17:59 	Setting Bandwidth: 1000000
2021/02/06 18:17:59 	SetTunerBw 1000000 Successful
2021/02/06 18:17:59 	ResetBuffer Successful
2021/02/06 18:17:59 	GetFreqCorrection: 0
2021/02/06 18:17:59 	SetFreqCorrection 0 Successful
2021/02/06 18:17:59 Entered UAT read() ...
2021/02/06 18:18:01 setting system time from 20210206 18:18:01.612 to: '20210206 18:18:01.400 UTC'
2021/02/06 18:18:01 Time set from GPS. Current time is 2021-02-06 18:18:01.400971614 +0000 GMT m=+4.352753436
2021/02/06 18:18:01 GPS detected: u-blox NMEA position message seen.
2021/02/06 18:18:01 setting system time from 20210206 18:18:01.669 to: '20210206 18:18:01.900 UTC'
2021/02/06 18:18:01 Time set from GPS. Current time is 2021-02-06 18:18:01.900937812 +0000 GMT m=+4.632908592

b3nn0, 2021-02-06:
Sorry, never tested with a BME, only BMP, but AFAIK it should work just fine.
I'm afraid I can't really help you with that.
You could potentially verify with a different software, e.g. something like this:
https://bitbucket.org/MattHawkinsUK/rpispy-misc/raw/master/python/bme280.py
if the Stratux software, or some hardware related issues are the cause.

Quax-der-Bruchpilot, 2021-02-06:
The BME and the BMP are very similar so I agree it should work. On my sensor it says “BME/BME” so I assume they are pretty much the same. Is there anything else that yo7 changed when you used the sensor? I will try with the other software that you recommended. Just to clarify: you used the I2C bus as well for the sensor, right?

b3nn0, 2021-02-06:
Yes, the sensors is read via i2c - i didn't change anything except for enabling it in the settings and connecting it.
I've not tested that python script to read the BME, it was just a quick google search that brought it up. My recommendation is just to test the complete hardware setup that you have with a different software - feel free to test the python script I linked, or anything else that the internet might bring up.

Quax-der-Bruchpilot, 2021-02-06:
Thanks for your help. Does the stratux software enable the I2C bus when I flash it? I just read that it needs to be enabled upon start up. The raspberry is brand new, so I guess this is something that needs to be done.

b3nn0, 2021-02-06:
As long as you flashed the normal Stratux image (and didn't compile yourself and install manually or something), then yes:
https://github.com/b3nn0/stratux/blob/master/image/config.txt

Quax-der-Bruchpilot, 2021-02-06:
Okay, thank you. I just flashed the latest image, nothing else. 

I need to run the sensor with the normal script that you send and see if it works. I might report back to see if I was able to fix it. Btw, I noticed that when I click on restart application the ALT light turns green for about 5 sec before it turns red again

Quax-der-Bruchpilot, 2021-02-06:
Okay, I just found something out: On the radar it says that it has received arp-Alt (see screenshot) but on the GPS/ AHRS tab it still is red. Please note that I unplugged all SDRs and GPS for this test.

b3nn0, 2021-02-06:
Can't post screenshots via mail reply. If you want to add a screenshot, you have to comment on the github issue report.
I think you can ignore the green Baro indicator in the radar page. I see it as well when no Baro is connected.. probably a bug.

Quax-der-Bruchpilot, 2021-02-06:
Well then I am back to square one :D Will keep trying tough

Quax-der-Bruchpilot, 2021-02-06:
Would you be able to provide an image with the setup that you used when you tested the compatibility with the BME? That would help a lot.

lyusupov, 2021-02-07:
```
/*=========================================================================
    I2C ADDRESS/BITS/SETTINGS
    -----------------------------------------------------------------------*/
    #define BMx280_ADDRESS                (0x77)
    #define BMx280_ADDRESS_ALT            (0x76) /* SA0 is NC */
    #define BMP280_CHIPID                 (0x58)
    #define BME280_CHIPID                 (0x60)
```

b3nn0, 2021-02-07:
Now that explains it. Apparently the BME wasn't ever supported (or used a different ID in the past?)
From Stratux code:
[image]

Sadly, this is part of the goflying module, not Stratux main module. Always hoped that I could get away without forking that one as well..

Quax-der-Bruchpilot, 2021-02-07:
Okay, I am not sure what I am looking at here. I see that Address1 is the 76 that I am using for my BME. The only reason I am using the BME is That the Stratux AHRS Module is sold out everywhere and also quite expensive since I am only interested in the pressure data from it. What do you suggest I should do now?

b3nn0, 2021-02-07:
I've thought about it for a bit, and I think I don't want to fork the "goflying" module that implements the pressure sensor/AHRS stuff.
So final conclusion:
<b>Stratux does not support the BME280, only the BMP280</b>

So you have 2 options:
1. Get a BMP280 instead of a BME280. It's like <2€ or something, so this is what I'd do personally. IMO it's not worth spending a lot of time for such a low priced component..
or
2. Change the code locally and compile Stratux yourself with BME280 support. But then you would need to do that in the future for each Stratux EU update. The general procedure would be something like
```
sudo -s
cd /root/stratux/goflying/bmp280/
nano bmp280.go
```
then replace `ChipID1 = 0x56` with `ChipID1 = 0x60` and save the file, then
```
cd /root/stratux
make && make install && stxrestart
```

b3nn0, 2021-02-07:
Ah, or 3: If you have a TTGO T-Motion or T-Beam with SoftRF or OGN Tracker firmware, you can just connect the BME280 to that one and replace your GPS with the T-Beam/T-Motion.
SoftRF and OGN Tracker both write the pressure altitude to their serial output and Stratux will use that.

Quax-der-Bruchpilot, 2021-02-07:
Thanks a lot for your detailed explanation. I have already ordered the TTGO T Beam, so I think this approach works best. The issue is now resolved and can be closed. Cheers

closed 2021-02-07 by b3nn0

kaelsaes, 2021-02-07:
I'm not 100% sure but I believe that BME280 is not supported by the OGN tracker software!

Quax-der-Bruchpilot, 2021-02-07:
Thanks, will get a BMP as well just to be sure.

Quax-der-Bruchpilot, 2021-02-07:
Just a quick update: Changing the Chip ID to 0x60 does not work unfortunately.

Quax-der-Bruchpilot, 2021-02-07:
I tried a few things and noticed, that in the goflying/bmp280 directive, there is a test folder with a file read_bmp280.go. When I execute it it displays the pressure and temperature data correctly. I changed the Chip ID to 0x60 for this and ran `make && make install && stxrestart.` I still don't see the baro pressure in the web interface tough. Do I need to adapt the your code somewhere as well? Seems to be that there is an interface issue somewhere.