stratux/stratux #104: ogn-rx-eu crashing

sternflyer, 2021-03-04:
1. Stratux version: 1.6r1-eu23
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: USB u-blox 8 GNSS receiver (NMEA protocol)

    AHRS
      - [ ] yes
      - [X] no

    power source: Wall Plug 5V 2A

    usb cable: Fixed to wall plug

3. EFB app and version: none

    EFB platform: none

    EFB hardware: none

4. Description of your issue:

I have just set up a rPi 3B with stratux. For testing reasons, I have it configured, that it is connected to my home WiFi (stratux is not providing an AP of it's own). 
At first it was working fine, but now I have the issue, that the process ogn-rx-eu is using up 100% Memory and crashing. When working, ogn-rx-eu is utilising ~140% CPU and ~7% MEM (values from "top"). When crashing, it's using 100% CPU (one full core) and memory goes up to 100% within about 30s. It then crashes and restarts. I have attached an excerpt from the log file. 

[stratux_ogn-rx-eu_crash.log](https://github.com/b3nn0/stratux/files/6084852/stratux_ogn-rx-eu_crash.log)

It is sitting indoors, so GPS reception is next to none.

Could it be, that it has to do with two different time sources (GPS an NTP). Since it is connected to my WiFi, it has access to the internet and hence raspbian is utilising NTP. In an earlier log I found, that ogn-rx-eu is crashing after the system time was updated due to GPS (look at the inconsistent time stamps).
[stratux_ogn-rx-eu_crash_gps_time.log](https://github.com/b3nn0/stratux/files/6084901/stratux_ogn-rx-eu_crash_gps_time.log)

Any suggestions?
Thanks!

b3nn0, 2021-03-04:
Did you run apt-get upgrade or something like that? Because you should not.
The packages that ship with Stratux are tested well.
Most notably, the latest Kernel update that is distributed with RaspiOS will cause OOMs for ogn-rx-eu.
We have no solution for this yet, except manually compiling the latest version of librtlsdr (the osmocom one).
It seems to be a weird bug somewhere in the kernel, libusb or librtlsdr. Not sure which one it is - it's extremely hard to debug.

closed 2021-03-04 by b3nn0

b3nn0, 2021-03-04:
FYI - and for anybody interested:
This seems to be the bug that causes the issues:
https://github.com/raspberrypi/linux/issues/3818
Due to the slow performance, ogn-rx-eu's demodulation buffer runs full because the demodulators can't keep up.