stratux/stratux #107: Where is the configuration menu for OGN tracker ?

MisterT2, 2021-03-07:
1. Stratux version: 1.6r1-eu023
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: VK-162 G-Mouse

    AHRS
      - [ ] yes
      - [X] no

    power source: external

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

I installed my TTGO T-Beam on my Stratux-EU 
I successfully upgraded its firmware directly from the Raspberry prompt, and rebooted everything.

But now, how do I configure the OGN-tracker ?
According to the release notes page (and this [picture](https://user-images.githubusercontent.com/1858945/102698336-df7e4f00-423c-11eb-9538-4ab75e74feb5.png)), I was expecting a minimum of configuration options in the "Settings" menu (or elsewhere), but I don't have anything...

Am I wrong in my expectations ? Did I miss something ? Or is there an issue ?

Thanks !

b3nn0, 2021-03-08:
That part on the Settings page only comes up when Stratux detects that there is the OGN Tracker firmware running. So make sure that
- You cleared your browser cache after installing the update
- On the Status page, it should say something like GPS: OGN Tracker.

Above you wrote GPS type: VK-162 -- I hope you disconnected that GPS? It can only be supplied by one source (which will then be the T-Beam).

If you don't want to use the T-Beam as GPS Source, you can still unplug the GPS temporarily to configure OGN Tracker, then plug it back in. In this case the T-Beam will only be used for transmission.

MisterT2, 2021-03-08:
Indeed, that was the "issue". Thanks a lot !

When unpluggin the GPS, Stratux is properly discovering the OGN Tracker and I can configure it. Unfortunately, its GPS is weak (at least indoor), and I'd rather use my external one.
So:
1. Unplug the external GPS
2. Configure tracker
3. Plug external GPS
All is back to "normal" (if I may say so)

Sorry for the confusion.

P.S.: it could be good to explain this "trick" somewhere (I could not find it) in the Wiki.
And may be also adding some advises about the OGN Tracker configuration... Personally, I decided to put the same Hex identifier in the Tracker as my Transponder code (recorded in Stratux configuration also). I don't know if I'm right ???

b3nn0, 2021-03-08:
Feel free to create a Wiki article if you want to - I'll proof read it and fix it if there is something wrong.
The "normal" way would be to remove the GPS and use the OGN Tracker instead. The GPS Chip on these T-Beams is decent, but the antenna is crap (many people just replace the antenna, since it uses a standard IPEX connector).
The build guides kinda imply that as well, since they don't include an additional GPS, but only the T-Beam.

Regarding configuration:
If you mostly fly one airplane, I'd recommend using your transponder HEX and setting address type to ICAO, yes.
The other options are mostly for aircraft without transponders or for people who switch aircraft frequently and don't want to reconfigure every time.

closed 2021-03-08 by b3nn0