stratux/stratux #115: AHRS sensor GY-91 not working with eu23

flyingzombies, 2021-05-05:
1. Stratux version: 1.6r1-eu023
2. Stratux config:

    SDR
      - [ ] single
      - [ X ] dual

    GPS
      - [ X ] yes
      - [ ] no
      type:

    AHRS
      - [ X ] yes
      - [ ] no

    power source: 2.0A 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

The GY-91 I installed only delivers baro-data, but no attitude (https://www.amazon.de/gp/product/B07DJ5L1YW/ref=ppx_yo_dt_b_asin_image_o00_s01?ie=UTF8&psc=1).
I tried to follow the hardware-approach on: https://github.com/b3nn0/stratux/issues/73
Stratus log shows as follows:

2020/05/27 10:39:28 stats [started: 14 minutes ago]
2020/05/27 10:39:28  - Disk bytes used = 2.3 GB (1.8 %), Disk bytes free = 123 GB (98.2 %)
2020/05/27 10:39:28  - CPUTemp=56.92 [48.31 - 58.53] deg C, MemStats.Alloc=1.8 MB, MemStats.Sys=11 MB, totalNetworkMessagesSent=1,427
2020/05/27 10:39:28  - UAT/min 0/0 [maxSS=0.00%], ES/min 1/1, Total traffic targets tracked=0
2020/05/27 10:39:28  - Network data messages sent: 1248 total, 1248 nonqueueable.  Network data bytes sent: 28697 total, 28697 nonqueueable.
2020/05/27 10:39:28  - Last GPS fix: 14 minutes ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 999999.00 m
2020/05/27 10:39:28  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 999999 m
2020/05/27 10:39:28  - Mode-S Distance factors (<5000, <10000, >10000): 2500.000000, 2800.000000, 3000.000000
2020/05/27 10:39:28 - Last IMU read: 14 minutes ago, Last BMP read: now
2020/05/27 10:39:32 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:36 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:40 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:44 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:46 signal caught: terminated - shutting down.
2020/05/27 10:39:47 Entered UAT shutdown() ...
2020/05/27 10:39:47 UAT shutdown(): calling u.wg.Wait() ...
2020/05/27 10:39:47 UAT read(): shutdown msg received...
2020/05/27 10:39:47 UAT shutdown(): u.wg.Wait() returned...
2020/05/27 10:39:47 UAT shutdown(): closing device ...
2020/05/27 10:39:47 UAT shutdown() complete ...
2020/05/27 10:39:47 Entered ES shutdown() ...
2020/05/27 10:39:47 ES shutdown(): calling e.wg.Wait() ...
2020/05/27 10:39:48 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:48 ES read(): shutdown msg received, calling cmd.Process.Kill() ...
2020/05/27 10:39:48 kill successful...
2020/05/27 10:39:48 ES shutdown() complete ...
2020/05/27 10:05:48  - stratux-pre-start.sh - Running Stratux Updater Script.
2020/05/27 10:05:48  - stratux-pre-start.sh - Exited without updating anything...
2020/05/27 10:39:48 Stratux v1.6r1-eu023 (fbda2940a4200d6a6fcf56f9e18e3e46bf6ee1f2) starting.
2020/05/27 10:39:48 read in settings.
2020/05/27 10:39:48 Developer mode set
2020/05/27 10:39:48 client connected: 192.168.10.10:4000 (Moritz).
2020/05/27 10:39:48 client connected: 192.168.10.10:2000 (Moritz).
2020/05/27 10:39:48 client connected: 192.168.10.10:49002 (Moritz).
2020/05/27 10:39:48 Added critical system error: System uptime 934 seconds. Daemon was restarted.

2020/05/27 10:39:48 Entered uatReader() ...
2020/05/27 10:39:50 ===== ES Device Serial: stx:1090:0 PPM 0 =====
2020/05/27 10:39:50 Entered ES read() ...
2020/05/27 10:39:50 Executed /usr/bin/dump1090 successfully...
2020/05/27 10:39:50 ===== UAT Device Name  : Generic RTL2832U OEM =====
2020/05/27 10:39:50 ===== UAT Device Serial: stx:978:-18=====
Found Rafael Micro R820T tuner
2020/05/27 10:39:50    GetTunerType: RTLSDR_TUNER_R820T
2020/05/27 10:39:50    SetTunerGainMode Successful
2020/05/27 10:39:50    SetTunerGain Successful
2020/05/27 10:39:50    GetTunerGain: 480
Exact sample rate is: 2083334.141630 Hz
[R82XX] PLL not locked!
2020/05/27 10:39:50    SetSampleRate - rate: 2083334
2020/05/27 10:39:50    GetSampleRate: 2083334
2020/05/27 10:39:50    GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2020/05/27 10:39:50    SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2020/05/27 10:39:50    SetCenterFreq 978MHz Successful
2020/05/27 10:39:50    GetCenterFreq: 978000000
2020/05/27 10:39:50    Setting Bandwidth: 1000000
2020/05/27 10:39:50    SetTunerBw 1000000 Successful
2020/05/27 10:39:50    ResetBuffer Successful
2020/05/27 10:39:50    GetFreqCorrection: 0
2020/05/27 10:39:50    SetFreqCorrection -18 Successful
2020/05/27 10:39:50 Entered UAT read() ...
2020/05/27 10:39:53 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:39:56 Could not identify MPU. v=c7, v2=70.
2020/05/27 10:40:00 Could not identify MPU. v=c7, v2=70.

i2cdetect -y 1 gives the same results as in issue 73:
If the SDO/SAO-Pin is not connected:
**0x68 and 0x76**
If it is connected to 3V3:
**0x69 and 0x77**
After connecting it to 3V3, the Baro-Alt in the AHRS-Tab does not update anymore and stratux.log is full of:
2020/05/27 10:43:00 BMP280 Warning: error reading sensor data: BMP280 Error reading from F7: %!s(<nil>)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

b3nn0, 2021-05-06:
Addresses 0x68/0x76 are fine, so Stratux is detecting a gyro.
It then tries to read two different register values from 0x68 to determine if it is an MPU-9250 or ICM-20948 (which both have address 0x68).
However, the registers don't have one of the expected values, so Stratux isn't sure what is connected there.
For MPU9250/9255 it expects register 0x75 to read 0x71/0x73. Your chip however delivers 0x70.
From quick googling, this should be the case if you have an MPU-6500. Maybe the GY-91 changed chips because the MPU9250 is not produced any more?
Apparently, the MPU-6500 is basically the same, except that it doesn't feature a magnetometer (which - IIRC - we are not using anyway).

You could try changing the MPUREG_WHO_AM_I_VAL to 0x70 and check if it is compatible (or add a new variable and change the respective `else if`).

Your steps to recompile the Stratux code are correct, not sure where something went wrong..

flyingzombies, 2021-05-06:
I was wondering why make would only take 3s
Turns out my system time was behind some of the files in the stratux folder.
Make just spit out "Clock skew detected. Your build may be incomplete" and ignored my changes.
`find . -exec touch {} +` in the stratux directory did the trick by setting all the files dates to the current system time.
Running make again now took several minutes.

I changed the MPUREG_WHO_AM_I_VAL to 0x70 and it worked!!!
Thanks for the quick help! :)

b3nn0, 2021-05-06:
Strange. Anyway, I added 0x70 so it should work OOTB in future versions.
Usually, Stratux will set your clock correctly once it has a GPS fix, but I guess you're testing indoors without good GPS reception.

closed 2021-05-06 by b3nn0

flyingzombies, 2021-05-06:
Yeah I'm indoors, so that should be the problem.
Nevertheless, workaround works 👍 

Appreciate the help

McNugget6750, 2022-08-03:
I seem to have the same issue with a recently purchased GY-91. It reports a chipID of 0x75 at register 0x75.
[image]

Since that's the case, I also get a 
Could not identify MPU. v=c0, v2=75

I have this one: https://www.amazon.com/dp/B0B42GMG1R?psc=1&ref=ppx_yo2ov_dt_b_product_details

b3nn0, 2022-08-03:
https://github.com/b3nn0/stratux/discussions/182