stratux/stratux #117: No function since eu024

dau-01, 2021-05-30:
1. Stratux version: 1.6r1-eu024
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: BL1

    AHRS
      - [x] yes
      - [ ] no

    power source: EasyAcc PB600CB Powerbank

    usb cable: integrated

3. EFB N/A

4. Description of your issue:

I've been using the EU-Stratux for some months now and decided to update to the eu024 PREVIEW. Since then, I can neither get into the web interface nor do my EFB apps recognize the Stratux.

There is still a network with the ssid "stratux" being broadcast and I can connect to it, but that's about it.

I noticed the ACT LED showing constant green, which could indicate a microSD-Card-error. So I ordered a new microSD, flashed it - no improvement. I removed the SDRs, GPS and the fan - no improvement. I reflashed the microSD with eu023, same issue. I tried a few other versions and also a cyoung-stratux, nothing changed.

At this point I was pretty convinced the RPi (3B) was just borked, so I went to confirm this by flashing the microSD with RPi OS - and this works flawlessly. At least as far as I clicked around the desktop.

So now I'm pretty much out of ideas. Could the RPi still be done for and for some reason only Stratux has issues with that? Is there anything I can do to troubleshoot further?

b3nn0, 2021-05-30:
Hi,
I'm pretty sure this is not related to eu024 - only had positive feedback for now and it can't potentially break your PI so much that even older versions don't work any more - that's just not possible I think.

Regarding your issues:
A solid green LED means that the Stratux process is running, so this is pretty normal if **everything is ok**.
Your symptoms, however (Wifi up, but no web interface), would indicate exactly the opposite - PI/OS running just fine, but Stratux process not running.

Since the LED is indicating a running process, can you - if you didn't already - please verify that the web interface isn't reachable with a different client device either?

How familiar are you with ssh?
I'd recommend connecting via ssh (which should be possible - ssh is up if the AP is up).
Then check `htop` if the gen_gdl90 process is running - that's the main Stratux process.
Additionally `cd /var/log && cat stratux.log` to check if the Stratux process is/was logging something (i.e. is/was running).

However, since this happens with a completely clean Stratux image flash, I'm fairly convinced that this is more of a network communication issue between your client device and the Stratux..

dau-01, 2021-05-31:
Thanks! I could connect via SSH and everything was running as it should. So I turned back looking for network issues and it turns out deleting the "stratux" wifi profile on my client device solved the issue. Not sure why, never had to do that before, but anyway, it's working now.

closed 2021-05-31 by dau-01